sandeepmistry/arduino-nRF5 #288: add FT2232 to the programmers menu?

ec1oud, 2018-06-25:
I suppose there must be some other project somewhere which provides the json magic to add a DLP2232 board as a JTAG programmer in the Arduino Programmer: menu item, but I'm having trouble finding one so far; any clues?

ec1oud, 2018-06-25:
I guess this is a matter of adding to programmers.txt and adding an openocd cfg file.  So far I've been trying to follow https://www.allaboutcircuits.com/technical-articles/getting-started-with-openocd-using-ft2232h-adapter-for-swd-debugging/ so I copied the jlink lines in your programmers.txt and modified like this:

```
ft2232.name=FT2232
ft2232.communication=USB
ft2232.protocol=ft2232
ft2232.program.protocol=ft2232
ft2232.program.tool=openocd
ft2232.program.setup_command=transport select swd; set WORKAREASIZE 0;
```

where I guess one of those protocol lines will make it load an ft2232.cfg file right? (and what is the other one for?)  And then my ft2232.cfg file looks like

```
interface ftdi
ftdi_vid_pid 0x0403 0x6010
ftdi_channel 1
#ftdi_serial "FTZ7O8O0" 
transport select swd 
ftdi_layout_init 0x0018 0x05fb
ftdi_layout_signal SWD_EN -data 0
ftdi_layout_signal nSRST -data 0x0010
adapter_khz 1000
```

but I suppose that may be redundant because there are a bunch of config files under `~/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5/scripts/interface/ftdi` already (too bad there isn't one for the dlp-2232, but I guess they consider that too old to bother with, or something; maybe nobody has used it for jtag before?  I just happened to have a few of them lying around.)  There is ftdi/swd-resistor-hack.cfg (that hack is what I'm trying to use) but it only has two lines which are copied into my config file.

When I try on the command line, it doesn't give me errors, just sortof sits there, and when I telnet to port 4444 I get connected but there's no interaction, no prompt.

Well maybe it even worked, I'm not sure... Arduino says "Done flashing software device" and the output in the black log window below is

```
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-09-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 1000 kHz
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
```

I just figured I'd see more than that.

dlabun, 2018-06-25:
I've never heard of anyone using an FTDI chip as a SWD... For the purposes of this project it's not supported in any fashion. You may want to ask around on the OpenOCD forums or mailing list for help.

With that being said, it looks like you only flashed a SD... Did you try flashing a project?

ec1oud, 2018-06-28:
Yeah, I tried.

I thought that a 2232 was a fairly common way of doing JTAG.  I also have an OpenMoko debug board (but never used it); that also uses an FT2232D in this way (for bit-banging).

dlabun, 2018-06-28:
My only suggestion is to use one of the supported programmers... Personally I would recommend picking up a ST-Link as they're globally available from Digikey and Mouser for about $20.

dlabun, 2018-07-23:
Closing due to lack of activity

closed 2018-07-23 by dlabun

ec1oud, 2018-07-24:
The Blue Pill showed up, I turned it into a Black Magic Probe, and got it working.