sandeepmistry/arduino-nRF5 #289: ADC Reads Full-Scale

roobituk, 2018-06-27:
I'm using pin P0.05 (AIN3) as an analog input to read a simple 4k7/4k7 potential divider between 5V and GND (so, approximately 2.5V). No matter what I try, when doing an analogRead(PIN_A3) only ever returns a full-scale reading. I can check the pin with a multimeter and I can see that 2.5V (ish) is going into the pin.

I can happily swap to other analog input pins but the result is the same.

I've set the resolution as 14-bit and the reference as AR_INTERNAL. Any help would be massively appreciated.

dlabun, 2018-06-27:
The internal reference is 06VDC +/-... What did you set as a gain? If you use 1/6 you can get in an input range of 3.6VDC

dlabun, 2018-06-27:
As far as I know, there's no active work to make that configurable beyond changing the .c file.

closed 2018-07-03 by dlabun