sandeepmistry/arduino-nRF5 #291: I2C AND attachInterrupt current consumption in sleep mode

dots123, 2018-07-02:
I'm using a module with NRF52832 (as a Generic NRF52 board with RC oscillator and s132) without any conections, excepting SWD interface. I put it into sleep mode with __WFI(); and it consumes 2.6 uA.
When I add an attachInterrupt the current jumps to ~12 uA, which is fine for me.

But when I initialise my accelerometer (using Sparkfun's MMA8452Q library, but in sleep mode (~8 uA)) the current jumps to 450 uA, which is not ok.

Now, if I remove attachInterrupt and keep only the accelerometer, it consume only 2.6 uA (like in the begining).

It's normal to consume 450 uA when are both?

coltonottley, 2018-07-19:
Hey Dots123,

I know nothing about the Sandeepmistry implementation of the gpiote driver but I know in the Nordic SDK if you enable the high accuracy mode on the pin change interrupt you'll see exactly the 450 uA power consumption you described. Setting hi_accuracy to false will fix it.

Colton

dots123, 2018-07-23:
Hi Colton,

Quite good observation, thank you!
I'll search for a possible adaptation!

coltonottley, 2018-07-24:
I looked into the sandeepmistry  WInterrupts.c file for the attachInterrupt function and it does look like they only implemented the hi_accu version of the nRF52 pin interrupt.  Also, with the NRF52 if you don't disable the TWI interface before sleeping you will get increased power consumption. I always disable the TWI when I'm not using it. You can probably use Wire.end() to achieve this.

dots123, 2018-07-24:
Colton, 

Thanks a lot for guiding!

I've tried with Wire.endTransmission() but nothing important happened.
But your idea of disabling was good. I've searched for an adaptation and I've discovered this:
`NRF_TWI1->ENABLE=TWI_ENABLE_ENABLE_Disabled << TWI_ENABLE_ENABLE_Pos;`
But this alone doesn't work because of a nrf52 error on chip. You can read about [Errata 89](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.Rev1.errata%2Fanomaly_832_89.html&cp=2_2_1_0_1_26)

So I've added:
```
*(volatile uint32_t *)0x40004FFC = 0;
*(volatile uint32_t *)0x40004FFC;
*(volatile uint32_t *)0x40004FFC = 1;
```
and it worked great.

Thank you again!

closed 2018-07-24 by dots123

nikolac, 2020-12-01:
Where is the "right" place to put this code?
```
*(volatile uint32_t *)0x40004FFC = 0;
*(volatile uint32_t *)0x40004FFC;
*(volatile uint32_t *)0x40004FFC = 1;
```
I assume it should live in TwoWire:endTransmission() of Wire_nRF2.cpp?

dots123, 2020-12-02:
Before going to sleep, in loop. Like this:

 NRF_TWI1->ENABLE = TWI_ENABLE_ENABLE_Disabled << TWI_ENABLE_ENABLE_Pos;
 NRF_TWI0->ENABLE = TWI_ENABLE_ENABLE_Disabled << TWI_ENABLE_ENABLE_Pos;

//correct errata 89
*(volatile uint32_t *)0x40004FFC = 0;
*(volatile uint32_t *)0x40004FFC;
*(volatile uint32_t *)0x40004FFC = 1;

And you need to restart it with Wire.begin(); after exit from sleep, if you need to use I2C again.

KEranda, 2021-01-25:
@dots123 did you ever figure out why when you attachInterrupt the current jumps to ~12 uA?
I'm having a similar issue where its ~3 uA before and ~9 uA? after attaching.

dots123, 2021-01-28:
Nope.

drmpf, 2022-02-25:
Does your module have a crystal?  I am trying to get a bare module without a crystal to run, but no go so far.
#477

muhdRiz07, 2022-10-14:
@dots123 hi, im using nrf52832 controller and i put into sleep mode using _wfi(). The current consumption in sleep mode is 93uA but when i add attachInterrupt the current jumps to 560uA. As u mentioned i added the below code before going to sleep but still the current consumption is 560uA

`NRF_TWI1->ENABLE = TWI_ENABLE_ENABLE_Disabled << TWI_ENABLE_ENABLE_Pos;
NRF_TWI0->ENABLE = TWI_ENABLE_ENABLE_Disabled << TWI_ENABLE_ENABLE_Pos;

//correct errata 89
*(volatile uint32_t *)0x40004FFC = 0;
*(volatile uint32_t *)0x40004FFC;
*(volatile uint32_t *)0x40004FFC = `1;