sandeepmistry/arduino-nRF5 #296: attachInterrupt() not working -> stops BLE Advertise

ericlangel, 2018-08-01:
Hello,

i'am trying to imlement an GPIO Interrupt.
See Arduino Low Power:
https://github.com/mristau/Arduino_nRF5x_lowPower/blob/master/examples/power_off/power_off.ino

But the attachInterrupt() function didn't work. The Handler never starts.
And the NRF52 stops advertising. without attachInterrupt() the Ddevive advertise.

Pin P0.13 -> Arduino Generic nRF52

pinMode(13, INPUT);
attachInterrupt(digitalPinToInterrupt(13), testHandler, RISING);

void testHandler() {
    Serial.println("Interrupt");    
}

Hope someone can Help me.
I'am trying to move from RFDuino ‎RFD22301 (NRF51822) to an NRF52832.
Everything works, except the sleep Mode.

Greetings Eric

dlabun, 2018-08-01:
If you are using a Soft Device I do not believe the standard Arduino interrupts will work. The soft device does take control of a lot of interrupts to allow operation of the Bluetooth modem. Check with Nordic documentation to see if it's possible for user code to use the interrupts with a soft device.

ericlangel, 2018-08-02:
ok, i will check the dokumentation.

so without using bluetooth (softdevice), the interrupt handler should work?

ericlangel, 2018-08-03:
i got want i wanted
i can set the NRF to system off and wake it with a pin.
i doesn't need the attach interrupt

But, in system off (60µA) i have no advertise and i can't connect with BLE
if i wake it up every 500ms i got advertising, but i cant connect.
Sure...system off = non conectable

But the mode low Power has a current of 5mA! 
Is there a Power mode with advertising+connect+arround 0-20µA?

Priyanshi1106, 2018-08-09:
Hi, 

Any solution to the original problem? Is there any way to use a GPIO interrupt with BLE?
I am using an MPU9250 and am trying to get it to wake on interrupt but the BLE advertising stops when I do attachInterrupt(). I have tried giving the GPIO interrupt a lower priority (higher priority number), but it still doesn't work. 

Thanks,
Priyanshi

ericlangel, 2018-08-10:
Hi Priyanshi,

i didn't use the attachInterrupt() function.
my next try is to use the RedBear Library. But i had no time to test it.

Which sleep mode do you want to use? System on or off?
could you get a wakeup from  system_on_sleep with a GPIO?

ericlangel, 2018-08-17:
i solved all my problems with sleep/BLE/external interrupt

removed the sandeepmistry lib and switch to the Redbear Nano V2 Lib-> works perfect!

closed 2018-08-17 by ericlangel

Priyanshi1106, 2018-08-17:
Hi Eric, 
I am not able to get the wire library working on RedBear Nano and so I'm using sandeepmistry lib. I got around the attachInterrupt problem by enabling it only after my device is connected. However, I wanted to know if there was a way of doing it without the workaround.

ericlangel, 2018-08-20:
n2k this workaround.

Wire library of redbear works fine in my application. did you started an issue?

Priyanshi1106, 2018-08-22:
Yes, I did. 

The workaround is just that instead of calling attachInterrupt in setup(), I call it in loop() after my ble device is connected.

hannesweisbach, 2021-06-06:
I believe the call to `sd_app_evt_wait();` should be followed by a call to `sd_nvic_ClearPendingIRQ(SWI2_IRQn);`, like so:

```
void loop() {
  Serial.println("Loop");
  sd_power_mode_set(NRF_POWER_MODE_LOWPWR);
  sd_app_evt_wait();
}
```
Otherwise, the (software) interrupt flag is still set and `sd_app_evt_wait();` will return immediately. The call to `delay` is superfluous.

jackmax, 2023-03-04:
The actual problem is that the GPIO interrupt priority is set to 2. According to Nordic Semi the S130 SoftDevice uses priorities 0 and 2, so the application should use priorities 1 and 3: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s130.sds%2Fdita%2Fsoftdevices%2Fs130%2Fprocessor_avail_interrupt_latency%2Fexception_mgmt_sd.html&cp=3_7_2_0_15_1

Setting priority to 1 or 3 in [this file](https://github.com/sandeepmistry/arduino-nRF5/blob/5c22636bb588d9afc43fec091dc0807a43f164bb/cores/nRF5/WInterrupts.c#L45) makes GPIO interrupts work. Setting it to 0 or 2 makes BLE interrupts (and serial port interrupts in my testing) stop working.