sandeepmistry/arduino-nRF5 #298: add GPIO Pin Mapping to readme?

austinfennacy, 2018-08-09:
First of all, thanks to all the contributors for this core, it's very helpful.

So I just finished spending the better part of my day figuring out how analog/digital pins in code (A1 or 1) map to the pin names on the datasheet (P0.##). I saw `C:\...\arduino-nRF5\variants\nRF52DK\variant.h` contained definitions for how certain constants map to pin numbers, such as `#define PIN_SERIAL_RX       (24)`, but I'm not sure what `(24)` refers to and learned it definitely wasn't P0.24.
I eventually found the GPIO Pin Mapping for the nRF52DK I'm using in the DK datasheet (attached below), and that cleared up all my confusion. Not sure if these mappings are specific to my module, I couldn't find anything clearer than "Section 4 Pin assignments" in the actual nRF52832 datasheet and while that has analog input mappings it omits digital pin mappings (e.g. D0 is mapped to P0.11).

How do most people learn about these pin mappings? If these pin mappings are consistent throughout most modules, could that be included in the readme? Or if not, perhaps something pointing people in the right direction for their specific module? I spent the longest time thinking my code wasn't uploading or somehow couldn't connect to the board.

Just hoping to help with the learning curve for future new users.

[PCA100~1.PDF](https://github.com/sandeepmistry/arduino-nRF5/files/2272864/PCA100.1.PDF)

dlabun, 2018-08-09:
The pin mapping is specific to each module. Most user's shouldn't need to know anything about the pin mapping as what's printed on the board should match the IDE. 

You do have a perfectly valid suggestion as the nRF DKs are the exception, Nordic printed the MCU pin numbers instead of the Arudino numbers on the board. The nRF52 DK is designed to hold Arudino shields so you can cheat and just print out a copy of the pin out map for an Uno or Zero and it'll work with your board.

austinfennacy, 2018-08-09:
What bad luck, that would be a lot more intuitive for boards that have Arduino numbers printed on them. Good suggestion, that may be easier than keeping the datasheet open.

Is that worth noting under the nRF52DK for "Supported boards"?

Also haven't used GitHub for open source projects before, should I make the change and send a pull request myself if I think that's something worth updating or is it poor form to contribute something so minor to a project? lol

dlabun, 2018-08-09:
You're more than welcome to do a PR for the note... I kinda doubt anybody else will do it.