sandeepmistry/arduino-nRF5 #301: Makerdiary.com nRF52832-MDK supported by generic device

w8abz, 2018-08-11:
All,
I was able to get the "generic nRF52" device to support the makerdiary.com nRF52832-MDK (which I got from Seeed Studio) by modifying the generic variant.h file to line up the various pins, especially the RGB LED and tx/rx pins for serial support.  The modified variant.h is listed below.  That was all I needed to modify to get everything I needed working.

Not so much an "issue" as hoping this helps someone else.

Sorry, I don't know the Arduino library/hardware structure well enough to create a whole new variant.

BTW, I hope you'll support the nRF52840-MDK soon, too!

Dale

`/*
  Copyright (c) 2014-2015 Arduino LLC.  All right reserved.
  Copyright (c) 2016 Sandeep Mistry All right reserved.
  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.
  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
  See the GNU Lesser General Public License for more details.
  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#ifndef _VARIANT_GENERIC_
#define _VARIANT_GENERIC_

/** Master clock frequency */
#ifdef NRF52
#define VARIANT_MCK       (64000000ul)
#else
#define VARIANT_MCK       (16000000ul)
#endif

/*----------------------------------------------------------------------------
 *        Headers
 *----------------------------------------------------------------------------*/

#include "WVariant.h"

#ifdef __cplusplus
extern "C"
{
#endif // __cplusplus

// Number of pins defined in PinDescription array
#define PINS_COUNT           (32u)
#define NUM_DIGITAL_PINS     (32u)
#define NUM_ANALOG_INPUTS    (8u)
#define NUM_ANALOG_OUTPUTS   (0u)

// LEDs
#define PIN_LEDG               (22)   // P0.22
#define PIN_LEDR               (23)   // P0.23
#define PIN_LEDB               (24)   // P0.24
#define LED_BUILTIN            PIN_LEDG

/*
 * Analog pins
 */
#define PIN_A0               (2)  // P0.02
#define PIN_A1               (3)  // P0.03
#define PIN_A2               (4)  // P0.04
#define PIN_A3               (5)  // P0.05
#define PIN_A4               (28) // P0.28
#define PIN_A5               (29) // P0.29
#define PIN_A6               (30) // P0.30
#define PIN_A7               (31) // P0.31

static const uint8_t A0  = PIN_A0 ;
static const uint8_t A1  = PIN_A1 ;
static const uint8_t A2  = PIN_A2 ;
static const uint8_t A3  = PIN_A3 ;
static const uint8_t A4  = PIN_A4 ;
static const uint8_t A5  = PIN_A5 ;
static const uint8_t A6  = PIN_A6 ;
static const uint8_t A7  = PIN_A7 ;
#ifdef NRF52
#define ADC_RESOLUTION    14
#else
#define ADC_RESOLUTION    10
#endif

/*
 * Serial interfaces
 */
// Serial
#define PIN_SERIAL_RX       (19) // P0.19
#define PIN_SERIAL_TX       (20) // P0.20

/*
 * SPI Interfaces
 */
#define SPI_INTERFACES_COUNT 1

#define PIN_SPI_MISO         (11) // P0.11
#define PIN_SPI_MOSI         (12) // P0.12
#define PIN_SPI_SCK          (13) // P0.13

static const uint8_t SS   = (14) ;  // P0.14
static const uint8_t MOSI = PIN_SPI_MOSI ;
static const uint8_t MISO = PIN_SPI_MISO ;
static const uint8_t SCK  = PIN_SPI_SCK ;

/*
 * Wire Interfaces
 */
#define WIRE_INTERFACES_COUNT 1

#define PIN_WIRE_SDA         (15u) // P0.15
#define PIN_WIRE_SCL         (16u) // P0.16

static const uint8_t SDA = PIN_WIRE_SDA;
static const uint8_t SCL = PIN_WIRE_SCL;

#ifdef __cplusplus
}
#endif

#endif
`

w8abz, 2018-08-11:
Sorry, tried to include the code in a "code" markup (markdown) but it didn't take.   And the path to the folder with the variant.h file I modified is:

C:\Users\myusername\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.5.1\variants\Generic

Also, thanks, Sandeep! I've been using this library with about a half dozen different devices.  This library and the BLEPeripheral library have been very useful!

Dale

sandeepmistry, 2018-08-12:
@w8abz we'd welcome a work in progress pull request for this, and can help you on it if you open one. Just ask some questions, examples of creating a new variant can be found in:

* https://github.com/sandeepmistry/arduino-nRF5/pull/259
* https://github.com/sandeepmistry/arduino-nRF5/pull/219
* https://github.com/sandeepmistry/arduino-nRF5/pull/95

w8abz, 2018-08-27:
Sandeep and all,

I was able to modify the boards.txt file and add this as a separate variant "makerdiary nRF52832-MDK" (rather than using the "Generic nRF52832" variant with only a modified variant.h file) on my Arduino IDE install.  I'm sorry I haven't had time to dig in and do the pull request.  Maybe this weekend...  But the new variant works nicely, and the pull mods should go pretty quickly once I have a chance to sit down and work my way through the exact steps I went through and the structure.

Thanks for the guidance so far!  Let me know where I mess up -- I need to learn git anyway!

Also, I now have a couple of the makerdiary nRF52840-MDK boards, so I hope I can help with adding support for the 52840 to this library.

Dale

w8abz, 2018-09-05:
Sandeep and all,

Okay, I'm learning as I go, and I hope I did this appropriately.   If not, I'll gracefully -- and gratefully -- accept corrections and hand-holding!

First, I did a fork from this repository to w8abz / Arduino-nRF5.
Then, I created a branch called  makerdiary-nrf52832-mdk.
Then, I created a new folder in the  variants  folder called nRF52832-MDK and added the three files that need to be there:   variant.h (modified for this board), and  pins_arduino.h  and  variant.cpp, which are unmodified copies of the ones for the Generic nRf52832 variant.
Then, I edited the boards.txt file (a couple folders up the hierarchy) to add the lines for this board, after the last nRF52832 variant and right before the set of nRF51822 variants.   (I hope I got the folder references here to match the folder name above!)
I committed all those changes, and I tried to write reasonable commit statements about what I had changed.

So, at this point, I think it's ready for folks to try out.  Am I coming up on being ready to do the pull request?

One thing I forgot to do is include the Issue number (301) in my commit statements.  Sorry.

Dale, W8ABZ

sandeepmistry, 2018-09-06:
> So, at this point, I think it's ready for folks to try out. Am I coming up on being ready to do the pull request?

Sounds like it!

I'm not sure if any of the maintainers own one of the boards to try out, however we are happy to review your changes. Maybe someone else from the community can try them out.