sandeepmistry/arduino-nRF5 #302: v0.6 Release

dlabun, 2018-08-12:
We haven't had a release since February, leaving support for Calliope Mini in limbo since March. I am going to suggest August 31st for our next release pending any other new developments.

We also have a number of old PRs, several over a year old. I think we need to either merge them or kill them off.

jhmaloney, 2018-08-13:
I agree. I would like to get the Calliope Mini support out. I was just about to open an issue to suggest that very thing.

dlabun, 2018-08-19:
@sandeepmistry I can handle pushing out the next release, I picked August 31st on the assumption that I'd work on it that weekend.

sandeepmistry, 2018-08-29:
@dlabun great!

dlabun, 2018-08-31:
@sandeepmistry if you are good with it I am going to merge #297 and push out v0.6 tomorrow.

sandeepmistry, 2018-08-31:
@dlabun I've added a comment related to #297 in https://github.com/sandeepmistry/arduino-nRF5/issues/280#issuecomment-417814495.

However if you feel like #297 is still worth merging tomorrow, please go ahead.

dlabun, 2018-09-03:
v0.6 is now live... I'm closing this issue as Travis CI is passing successfully.

closed 2018-09-03 by dlabun

jhmaloney, 2018-09-03:
Thank you!

	-- John