sandeepmistry/arduino-nRF5 #303: What to Expect when connecting Smart Beacon to Arduino

rmetzner49, 2018-08-13:
First of all, thanks for the great work in developing the NRF5 variants.  I'm a little unclear what to look for after having good success with the ESP32.

Assume I have a USB to Serial Cable that can work at 3.3V and I have the TxD and RxD connected to P0.09 and P0.11, should I look for a diagnostic message on power up?  What does the built-in bootloader say?  Does it need the SWDIO and SWDCLK for anything?

Has anyone connected an NRF51822 to Aduino?

Thanks.
Bob Metzner

dlabun, 2018-08-14:
There’s no boot loader in use, the project strictly relies on external programmers. I think Adafruit uses a boot load, so don’t get confused by their instructions.

rmetzner49, 2018-08-14:
Well, that makes it a bit less attractive.  Seemed like a goof fit for my project but not so much now.

dlabun, 2018-08-14:
To each his own, but you’re the first person I know of to express a need for a boot loader.

rmetzner49, 2018-08-14:
Ouch!  I think you might be confusing "need" for convenience.  Why even go through all the work of porting Arduino to the device if you can't support it in the customary fashion.  Oh wait ...  someone already did that, AKA Adafruit.

dlabun, 2018-08-14:
Closing due to attempted trolling.

closed 2018-08-14 by dlabun