sandeepmistry/arduino-nRF5 #307: Connect RedBear Nano 2 Module

ELIVB, 2018-08-25:
Hello
I do not speak good English I hope you understand me.

I tried to search for answers online without success.
Can I connect the RedBear Nano 2 Module or another Module like Ebyte nrf52840 E73-2G4M08S1C Directly to MCU over SPI) like Adafruit feather Express M4? (CPU with more speed and more memory).

https://www.adafruit.com/product/3857

There are several components connected to my project, I want to build my Bluetooth ble profile, service and Characteristic on the main program (not inside Soc)
I know I'll need SPI, IRQ and RST pin.

Is it possible to do so?
If possible, is there any example you know how to do this?
Can you help?
Note: I work in Arduino environment
Thanks
Eli

dlabun, 2018-08-25:
No, what are you asking for is not possible as your Bluetooth code must be executed on the nRF chip itself.

ELIVB, 2018-08-26:
Hello and Thanks for your response.

There is the following product from  adafruit company.
https://learn.adafruit.com/introducing-the-adafruit-bluefruit-spi-breakout/introduction

This is a Soc system ARM Cortex M0 core running at 16MHz (nRF51822), If I understood correctly the program run in arduino and communicate to the BLE by IRQ pin.

am I wrong?

eli

dlabun, 2018-08-26:
That part requires a full SPI connection, not just a interrupt pin.

ELIVB, 2018-08-26:
Hello Friend

Sorry, but I did not understand what you mean, do you now think that nRF52840  can be  connected to arduino by SPI?

thanks for the effort
eli

dlabun, 2018-08-26:
I understood the question, you want to use the Bluetooth radio on the nRF52840 while having all of your Bluetooth related code on a separate processor.... That is not possible due to the fact that you must execute the Bluetooth code directly on the SoC. If you look at the Adafruit part you referenced it contains a custom application written by Adafruit that translates their "protocol" into actions on the Bluetooth radio.