sandeepmistry/arduino-nRF5 #308: digitalRead since it always returns LOW despite digitalWrite was called with HIGH value.

Navis-Raven, 2018-08-31:
There is a notable bug with nRF52-DK with arduino framwork, as you can see on github here (see the point 7), click on this link : https://github.com/platformio/platformio-docs/issues/33#issuecomment-416382889

The problematic code is here: https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html#writing-unit-tests

there is a problem with function digitalRead since it always returns LOW despite the fact that digitalWrite was called with HIGH value.

The issue is also declared in Nordic: https://devzone.nordicsemi.com/f/nordic-q-a/38028/there-is-a-notable-bug-in-nrf52-dk-with-arduino-framework

IDE: VSCODE with platformio,
SDK/Framwork: Arduino

dlabun, 2018-08-31:
Can you reproduce this issue in the Arudino IDE? This project is intended only for use with the Arduino IDE.

Navis-Raven, 2018-08-31:
I don't have UNITY (unit testing) in my arduino IDE...
Anyway this bug is independant from the IDE: platformIO uses arduino console commands to build project (the same as arduino IDE)

Navis-Raven, 2018-08-31:
If you have time, try to run a test with your nRF52-DK 

By reading after writing HIGH, and print this into your serial debug.

dlabun, 2018-08-31:
Unfortunately this will have to wait until I have time to dig into that part of the project.

Navis-Raven, 2018-08-31:
@sandeepmistry it's a bug, not a "help wanted" post.

dlabun, 2018-08-31:
It's nothing more than a flag indicating that help from others to locate and correct the issue would be appreciated. If you would prefer that we remove the help wanted flag I can certainly do that.

Navis-Raven, 2018-08-31:
Ok, I didn't understand this like that.

jandrassy, 2018-09-03:
why should digitalRead on an OUTPUT pin return something meaningful? or digitalWrite write something to an INPUT pin?

Navis-Raven, 2018-09-03:
@jandrassy I don't know but the platformIO website referenced it as an issue of the arduino framework. 

[image]

jandrassy, 2018-09-03:
https://arduino.stackexchange.com/questions/13588/read-output-pin-value-in-arduino-due

Navis-Raven, 2018-09-03:
@jandrassy  so in your opinion it's the PlatformIO documentation which is incorrect, isn't it ?

valeros, 2018-09-03:
Hi everyone! I'm not a specialist in the Arduino framework, but I thought it's reasonable to expect that `digitalRead` should return the current state of the pin? Anyway. I've just tested the following code with Uno board in Arduino IDE and it works like I expected:
```
void setup()
{
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.print("Pin value after HIGH: ");
  Serial.println(digitalRead(LED_BUILTIN));
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  Serial.print("Pin value after low: ");
  Serial.println(digitalRead(LED_BUILTIN));
  delay(1000);
}
```
And here is the output:
```

dlabun, 2018-09-03:
It looks like this is expected behavior as it's been reported with all of the official Arduino boards using ARM processors. It's noted in multiple places that the AVRs (and specifically the Uno) are able to read the output simply because of the way their registers are designed. Since this project is correctly reflecting the same behavior as the official Arduino Zero and Due cores I am going to say this is purely a documentation issue on PlatformIO's side.

Navis-Raven, 2018-09-03:
@valeros 
This is what @jandrassy was pointing
[image]

dlabun, 2018-09-05:
I’m closing this as not an issue with project but rather the PlatformIO documentation.

closed 2018-09-05 by dlabun