sandeepmistry/arduino-nRF5 #309: Port to Pin Definition Location

Deanasawr, 2018-09-04:
Hello there.

I'm making my own NRF52832-based board, and would like it to be programmable via the Arduino IDE.

I've already made and tested the board, and can upload and run sketches using a J-Link and the Nordic Semiconductor nRF52DK board definition. However, due to the formfactor of the NRF52 I used (BT832X), some of the ports are reserved, and needless to say the rest of the pin definitions don't quite match up to where I mapped them on the circuit board.

I managed to find and re-map the more crucial pins such as the I2C, UART, and LED pins using the Variants.CPP file and a little trial and error to obtain basic functionality, but I really need the rest of the pins eventually.  

tl;dr: does anyone know where the port to pin definitions is stored besides what's contained in variants.cpp?

dlabun, 2018-09-04:
variants.h also contains pin definition information and I believe between those two files that's the only place any information is stored.

sandeepmistry, 2018-09-06:
> tl;dr: does anyone know where the port to pin definitions is stored besides what's contained in variants.cpp?

The array inside `variant.cpp` is used to map an Arduino pin number (one passed to `pinMode(...)` etc. ) to the Nordic pin number. This index into the array is the Arduino pin number.

dlabun, 2018-09-11:
Closing as resolved since we haven't heard back from the OP

closed 2018-09-11 by dlabun