sandeepmistry/arduino-nRF5 #315: Support for NRF52840-DONGLE

knopserl, 2018-10-04:
Nordic offers the nRF52840 USB Dongle which is not only cost effective, but also small, versative and supports BLE 5.0 long range, ZigBee and is also very convinient to use.
It would be great for the community to get this also working with Arduino IDE and Platform.IO.
I have seen, that the older nRF USB dongle is already supported but can't be used.
Is ther a chance to get this great dongle supported inthe near future. Many people are hoping that this will be supported.
Or is there a good description/recipe what has to be done to get it supported?

dlabun, 2018-10-04:
The short answer is that there would need to be a major overhaul of this project to use the latest Nordic SDK in order to support the nRF52840. Since there's been little effort from the community to move to the latest SDK it's not likely that support for the nRF52840 will come any time soon.

knopserl, 2018-10-05:
a real pitty! a couple of enthusiast an major HW/SW contributer of the scene is waiting and hoping that it will be supported like me. It would be really,really worth to support it.
There is a big advantage of the 840 in multiple areas in comparison to the 832.
So I still hope that somebody who is capable and mercifull to help us. Arduino IDE and platform.io would benefit from it.

sandeepmistry, 2018-11-10:
Let's keep track of this in #255. Closing as duplicate.

closed 2018-11-10 by sandeepmistry

knopserl, 2018-11-10:
really a duplicat of #255? The other one is requesting nRF52810 which I think is miles away from a nRF52840. Here we take basically about support of SDK15.2+ and siftdevice s140+
Are you sure that this should be merged?

sandeepmistry, 2018-11-10:
Sorry, duplicate of #70.