sandeepmistry/arduino-nRF5 #318: Compilation Error for NRF52832 based RACK 8212 iTracker Board !!

fu-techiot, 2018-10-19:
**Hello There,**
i have RAK8212 NB IOT  itracker BOARD it has NRF52832 chip.
i also have j link ARM Emulator with SWD interface.

using this thing in ARDUINO IDE I already flashed "NRFSOFTDEVICE" successfully.

but when I'm going to compile and upload the code led_switch from library BLE PERIPHERAL, I got following error.
 ```
 In file included from C:\Users\Arana\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.6.0\cores\nRF5/Arduino.h:5:0,
 from C:\Users\Arana\AppData\Local\Temp\arduino_build_931224\sketch\led_switch.ino.cpp:1:
```

```
c:\users\arana\appdata\local\arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4\lib\gcc\arm-none-eabi\5.2.1\include\stdint.h:9:26: fatal error: stdint.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board iTracker.
```

everytime the compilation is terminated.
looks like something is missing (stdin.h is missing) !!
but dont know what is missing!!

fu-techiot, 2018-10-26:
@dlabun yeah i had already install nrf board package. 
but i'm still getting following error
"i'm not able to flash NRFSOFTDEVICE"
```
Downloading 'http://www.nordicsemi.com/eng/content/download/95151/1606944/file/s132_nrf52_2.0.0.zip' ...
.done
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
jaylink: Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
jaylink: Failed to open device: LIBUSB_ERROR_NOT_FOUND.
Error: No J-Link device found.

Error while flashing SoftDevice.
```
In ARDUINO IDE i'm selecting port of "J tag" as we have to flash s132 soft device is it right? 
or I  have to select COM port of RAK8212 board??
in NRF GO studio i can see board is connected then what will bw the issue?

dlabun, 2018-10-26:
Please read the install instructions in the readme, the error message indicates you didn’t install the USB driver.