sandeepmistry/arduino-nRF5 #320: Error: unable to find CMSIS-DAP device   Error: No Valid JTAG Interface Configured.

pfodApp, 2018-11-09:
I am using blackmagic V2.1 on Windows 7 with  arduino-nRF5 V0.6.0  to program a RedBear NanoV2 board.
I managed to load the softdevice S132 (after manually downloading and installing the .hex file)
But when I try to load a sketch I get this error

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

From the blackmagic I am connecting pins CLK and DIO as well as VREF and GND
the NanoV2 is powered via VIN /GND from a 5V source

dlabun, 2018-11-09:
Did you install the LibUSB driver per the readme?

pfodApp, 2018-11-09:
No mention of LibUSB in the readme for this project.
However as per you suggestion I downloaded LibUSB drivers for Win X86 from https://sourceforge.net/projects/libusb-win32/
and tried to install them as per https://www.pinguino.cc/download/doc/libusb-windows7.pdf
but found that they were already installed.  (also found libUSBK drivers installed)

I tried to run the mbed Windows Serial driver (again) but got this error message 
"The driver could not be installed. No mbed Microcontrollers found"
Tried plugging in the blackmagic board but got same error.
Tried plugging in the Redbear nano programming board and drivers installed successfully.

Then tried to download sketch again, using blackmagic and sandeep's IDE add on, but got same errors as above

(p.s. have previously run zadig when trying to use a CMSIS-DAP board to program the nano using redbear's addon)

dlabun, 2018-11-09:
Sorry, I meant Zadig and not LibUSB... When you have the BMP plugged in do you seen any COM ports listed for it or is it just not showing up at all? I am wondering if you have conflicting drivers installed.

pfodApp, 2018-11-10:
I understood from the readme that zadig was used for Segger JLink and was not needed for blackmagic so I have not used it this time around although I tried it previously when trying to use a CMSIS-DAP board.

When I first plugged in the blackmagic board it was identified as two Teensy com devices and would not load the softdevice.  However I fixed this by uninstalling the Teensy coms and then two unknow coms showed up BlackMagic GDB server and BlackMagic UART port,  with question marks.
Clicking on Update Driver software and navigating to the dir containing 
blackmagic.inf and
blackmagic_upgrade.inf
allowed me to install drivers and the question marks disappeared and now look like normal COMs BlackMagic GDB and BlackMagic UART.  Selecting the GDB com allowed me to install the softdevice as shown below

```
C:\Users\matthew\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-gdb -quiet -ex target extended-remote \\.\COM113 -ex monitor swdp_scan -ex attach 1 -ex monitor erase mass -ex echo \nErase finished!\n -ex detach -ex quit 
Remote debugging using \\.\COM113
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF52
Attaching to Remote target
0x0001b08e in ?? ()
erase..

Erase finished!
Detaching from program: , Remote target
C:\Users\..\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-gdb -quiet -cd C:\Users\..\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.6.0/cores/nRF5/SDK/components/softdevice/s132/hex/ -ex target extended-remote \\.\COM113 -ex monitor swdp_scan -ex attach 1 -ex monitor erase mass -ex load s132_nrf52_2.0.1_softdevice.hex -ex echo \nSoftdevice upload complete!\n -ex detach -ex quit 
Target voltage: 3.3V
Remote debugging using \\.\COM113
Available Targets:
No. Att Driver
 1      Nordic nRF52
Attaching to Remote target
0xfffffffe in ?? ()
erase..
Loading section .sec1, size 0x964 lma 0x0
Loading section .sec2, size 0xf000 lma 0x1000
Loading section .sec3, size 0xb6c8 lma 0x10000
Start address 0x0, load size 110636
Transfer rate: 19 KB/sec, 970 bytes/write.

Softdevice upload complete!
Detaching from program: , Remote target

```
But trying to then load a sketch gives the following errors

```
C:\Users\..\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/cmsis-dap.cfg -c transport select swd; -f target/nrf52.cfg -c program {{C:\Users\..\AppData\Local\Temp\arduino_build_124931/sketch_nov07a.ino.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

```

pfodApp, 2018-11-10:
Here is some gdb output that may help pinpoint the problem

```
(gdb) monitor jtag_scan
Target voltage: 3.3V
JTAG device scan failed!
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF52
(gdb) attach 1
Attaching to Remote target
warning: while parsing target description (at line 1): Target description specified unknown architecture "arm"
warning: Could not load XML target description; ignoring
0x00000000 in ?? ()
(gdb)
```
Is the XML target description the problem?

pfodApp, 2018-11-10:
I removed some path variables to prevent another gdb being found and tried again, but same error

pfodApp, 2018-11-10:
I went back and check that BlackMagic GDB was selected as the programmer.  But even with it ticked and shown in the tools menu cmsis-dap is being used to program hence the failure. 
It seems the addon does not honour the programmer setting when RedBear NanoV2 is selected.
p.s. sketch (with BLE code) loaded and ran successfully when Generic nRF52 selected but ofcourse the pin mappings are not correct.

dlabun, 2018-11-10:
I’m closing this now that the titled has been changed to fixed.

closed 2018-11-10 by dlabun