sandeepmistry/arduino-nRF5 #322: Generic nRF52 board Analog pins not correctly defined

drmpf, 2018-11-25:
The Generic nRF52 board variant defines the A0 to A5 as pins 1 to 6 and the mapping array g_ADigitalPinMap maps these to 1 to 6,  but the nRF52823 chip
has the mapping as described in
https://github.com/sandeepmistry/arduino-nRF5/issues/301

Generic nRF52 needs to be corrected to follow the pinouts in above issue

dlabun, 2018-11-25:
There’s nothing wrong as it’s a generic template and does not match any specific board on the market.

closed 2018-11-25 by dlabun

pfodApp, 2018-11-25:
Not really useful then is it.
I imagined "Generic nRF52" was for those boards that just connected the nRF52832 chip pins to pads on the edge of be board,  eg 
http://www.skylabmodule.com/nfcant-bluetooth-4-2-module-skb369/