sandeepmistry/arduino-nRF5 #323: v0.7 Release

dlabun, 2018-12-02:
I am currently looking at doing a v0.7 release next weekend as the last release for this year. Currently there's only two items to be included:

-Fix for soft devices not installing (#312)
-Add support for DWM1001 board (#319)

There doesn't appear to be anything else ready to merge in.

sandeepmistry, 2018-12-02:
@dlabun sounds good, #311 is looking close to being ready, but I don't think we should wait for it.

Beetix, 2019-06-27:
Hi,

Is there any news on this release?

Thanks

dlabun, 2019-06-27:
I could probably push out this release this weekend... Honestly nobody has been begging for it so I wasn't too worried about it.

Beetix, 2019-06-27:
That would be great, thank you.

Beetix, 2019-10-16:
Hi,

@dlabun Are you still planning on making the release?

Thanks

sandeepmistry, 2020-12-04:
I’ll be making a new core release this weekend /cc @jhmaloney

sandeepmistry, 2020-12-05:
The 0.7.0 release has been made!

https://github.com/sandeepmistry/arduino-nRF5/releases/tag/0.7.0

closed 2020-12-05 by sandeepmistry