sandeepmistry/arduino-nRF5 #324: DFU for nRF5 Core with Adafruit Bootloader?

curtpw, 2018-12-04:
1: [Adafruit bootloader ](https://github.com/adafruit/Adafruit_nRF52_Bootloader)supports DFU
2: [Adafruit nRF52 core](https://github.com/adafruit/Adafruit_nRF52_Arduino) is based on Sandeep's nRFx Arduino Core (this repo), and works with the aforementioned bootloader.
Q: Does anyone have knowledge or experience that would help determine if the Adafruit bootloader can be made to work with [nRF5 Core](https://github.com/sandeepmistry/arduino-nRF5)? If so, what are your thoughts?

NOTE: this question continues #32 and #240

jeremypoulter, 2018-12-04:
First of all even though the Adafruit core was originally based on this core they are now quite different, however I believe it should be possible. I think the biggest issue (this is off the top of my head) is that with the Adafruit setup, the softdevice is part of the bootloader rather that the application so you will need to change the build to not include the softdevice (if you use the softdevice) as well as find/build a bootloader with an appropriate softdevice. You will also need to change the base address of the application and copy across the code to package up the DFU .zip.  

That being said it is probably simpler to just build the application on the Adafruit core.

curtpw, 2018-12-04:
Thanks, I'll update this issue if I have any luck one way or another.

closed 2018-12-04 by curtpw