sandeepmistry/arduino-nRF5 #325: J-TAG upload error

ninanas98, 2018-12-14:
Hi there,
I am using a customized board with nrf51822. But when I want to upload the sketch to the board I get the following error:

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

The board I use is: http://www.holyiot.com/product/nordic-nrf51822-ble-4-0-module-lis2dh12-accelerometer-sensor/

dlabun, 2018-12-14:
Which programmer are you trying to use? From the looks of the error message you don't have one installed.

ninanas98, 2018-12-14:
I am using Arduino IDE 1.8.8.

Installed the boards from sandeepmistry.
Used Zadig as well so I have no clue where it goes wrong.

ninanas98, 2018-12-15:
To connect the board to the pc? I use a USB to TTL

dlabun, 2018-12-15:
That won’t work to program the board, you need to use a real programmer. J-Link, ST-Link and Black Magic Probe are all supported.

closed 2018-12-15 by dlabun

ninanas98, 2018-12-15:
I get this error when i used J-Link. With magic probe I can upload the sketch but get an error no headers files (.h)