sandeepmistry/arduino-nRF5 #328: 2.4GHz transceiver like nRF24L01+

HAMUZAO, 2018-12-21:
Hello dear,
I just wanted to ask if this arduino NRF core support the 2.4GHz transceiver communication mode (not BLE) like the nRF24L01+ !?

Thank you :)

dlabun, 2018-12-21:
Sorry, but it does not. I am not too familiar with the Non-BLE communications but I don’t think the nRF5 family has that feature any more.

HAMUZAO, 2018-12-24:
Its ok ^_^
Your actual work is already fantastic 👍  I just wanted to ask that question because i was looking for a 2.4Gh chip taht can be programmed by arduino & i found this topic & some nRF5 boards on sparkfun but they skipp talking about the RF side ! So i though its better to ask the magician by myself ^_^ .. I think that ur project will be a revolution if its well invested to turn nrf chips standalone & programmable via Arduino IDE without the need to use an external MCU, If that project can be extended to include the nRF24xxx & support multiple networking options that will be super cool ^_^

I wish if i can finance u but im super poor at this moment :( !! but I think u should take ur project to Kikstarter & in that place every little donation will be valuable  ;)

Hope to see my wish become true soon ^_^
Thank you

dlabun, 2018-12-30:
Closing as answered.

closed 2018-12-30 by dlabun