sandeepmistry/arduino-nRF5 #330: nRF51822 - Bluetooth-Modul does not receiving from central

Lutz95326, 2018-12-29:
Hello, i have installed the Nordic Semiconductor Boards in Arduino App (Version 1.8.8) with the sandeepmistry - library (per "https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json"). Many thanks for your great work!

I am able to flash the softdevice S130 and run the Blinking-Led-Example with no problems.

But when i try the BLEPeriperal-LED-Example. It isn´t work properly.
I can see my BLE-Device on my smartphone. I start nRF-Connect on smartpon, connect to the BLE-Device an can see all Infos of the BLE (Gerneric Access, Generic Attribute, Unknown Service) in smartphone an the Messages "Connect/Disconnect" on the UART.
But if i try to send a Byte to BLE-Device for switching the LED on or off the BLE-Device disconnects after a while without switching LED on/off.
I have also made Serial.println() in sketch and see, that Bytes from central (smartphone) not come into the sketch per "switchCharacteristic.written()".
Same error with "s110_nrf51_8.0.0_softdevice" and "s130_nrf51_2.0.1_softdevice".

Have anyone a note for me? - Thanks

Lutz95326, 2018-12-30:
Hello, 
the problem above i a problem on the Module:

The same program, the same softdevices, one module works the other not...

This nRF51822-Modul (QFAAH0 1821YP) works perfectly with Bluetooth.
This nRF51822-Modul (QFAAH0 1524GG) works not with Bluetooth. Received Bytes from central (smartphone) do not come in my software, after a while the BLE-Module disconnect Bluetooth. Any other examples without Bluetooth (blinking LED ...) are working.

I suppose there is a problem in the chip.

How can i be shure to buy modules that works in future?

Best regards
Horst Härtel - Software Developer-

dlabun, 2018-12-30:
You will need to contact the module manufacturer to find out why one of the modules is faulty.

closed 2018-12-30 by dlabun