sandeepmistry/arduino-nRF5 #331: Latest Nordic SoftDevice (v6.1.x)

bobr-rigado, 2019-01-09:
Are there plans to update to the newer SoftDevices, especially since they will be required in order to obtain an end-product declaration with the Bluetooth SIG?

dlabun, 2019-01-09:
Unfortunately there are currently no plans to move to a newer soft device. The issue is that the newer SDs require using the newer Nordic SDKs and that would require a substantial update to this project. If somebody wants to help update the project in order to support newer SDs I'm all for it and will gladly merge in the changes, I just don't have the resources to do it myself.

I think it was around SDK v19 that Nordic dropped support for the nRF51 series so we would have to figure out how handle having two different chip families requiring two different SDK versions.

dlabun, 2019-01-14:
Closing as answered

closed 2019-01-14 by dlabun