sandeepmistry/arduino-nRF5 #341: undefined reference to `analogWrite'

Benni77, 2019-02-07:
Hey guys,

tryin to use the analogWrite function but I always get the undefined reference error?! I don't really get the problem?!
Any Ideas?
Thanks :)

dlabun, 2019-02-07:
Is this the first time your are trying to use this project? It does sound like your installation may be corrupt.

Benni77, 2019-02-08:
Mh actually my whole projekt worked totally fine, till analogWrite came :D
I should try reinstalling that stuff. Maybe it's because I'm using some files from the adafruit Installation..

Benni77, 2019-02-08:
Okay found a solution. I got the SparkFun nRF52832 Breakout Board.

I had to change the analog write functions name and comment out the analogwriteresolution in the library?!
Don't really know why but its working like this.

dlabun, 2019-02-08:
You would have to check with Sparkfun as none of their board are currently supported by this project.

closed 2019-02-08 by dlabun