sandeepmistry/arduino-nRF5 #342: Update OpenOCD to allow removing nRF52 programming protection

drmpf, 2019-02-12:
nRF52 modules like SKYLAB SKB369 and others come pre-programmed and protected against re-programming.  
The current version of OpenOCD supplied with this package does not support the apreg cmd necessary to remove the chip protection.
Please update to OpenOCD-20181130 
[Instructions for removing the chip protection are given here](https://www.forward.com.au/pfod/BLE/LowPower/NanoReplacement/index.html#removing)

A menu item in Arduino would be nice also if you have time.

dlabun, 2019-02-12:
Any legit module should arrive in an unprogrammed, unprotected state... Have you contacted Skylab to see why you have modules that are protected?

drmpf, 2019-02-12:
All of the five modules, I have come pre-programmed with a heat rate service. 
great for testing operation but need to ERASEALL to reprogram.
The protection is a common micro chip feature to prevent reverse engineering the software.
https://www.homesmartmesh.com/nrf52/  uses a different module from another manufacturer and it needs to be un-protected also.
So this appears to be a common problem but easily overcome with the latest version of OpenOCD

dlabun, 2019-02-12:
Since the modules are supplied from the factory intentional with protection turned on there's no issue on our side per say and makes this essentially a duplicate of #106

closed 2019-02-12 by dlabun

drmpf, 2024-03-11:
nRF52 with build codes Gxx and later have chip protection enabled by default after each Pin reset
• Power or brownout reset
• Watchdog reset if not in Debug Interface Mode
• Wake from System OFF if not in Emulated System OFF
So the process of writing the softdevice separate from the sketch no longer works.
The solutions are
i) add extra code to prevent the auto chip protection see https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/working-with-the-nrf52-series-improved-approtect for the details
OR
ii) doing a full chip erase (which remove the protection) and then writing both the softdevice and the sketch .hex before restarting/reseting the chip.

My https://www.forward.com.au/pfod/BLE/LowPower_2022/index.html modification of this repository does that.  
https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/working-with-the-nrf52-series-improved-approtect