sandeepmistry/arduino-nRF5 #348: nRF51822: TWI data corruption & TWI hang up sometimes

Shreyas-R, 2019-03-18:
Guys, I'm using the library for one of my projects using generic nRF51822 (v3) module where I'm using BLE peripheral (BLE UART) to send some data through I2C to some external device. I took examples from your library & built my code on that. Now I'm facing two issues:

1) Sometimes I2C data sent is corrupt
2) Sometimes I2C data transmission hangs nRF51822 (with SCL or SDA always low)

I did some debug & here is what I found:
1) I'm sending multiple I2C data bytes after each interval & I found that some clock cycles or some data is not sent over I2C.
2) Slave has fixed frequency set & clock stretching disabled so it must be caused by master only (especially SCL always low)

After going through some Nordic doc & threads, I found that BLE peripheral has higher priority & can cause issue to TWI. 

1) What & where is the priority of TWI set in the library?
2) Is DMA enabled for TWI?
3) Is TWI intialised with blocking or non blocking mode in the library? Where can I change it?

dlabun, 2019-03-18:
I didn't write the I2C library so I may be incorrect on this, but as far as I know there's no priority set and DMA is not used. I do not know about blocking mode, you may want to review the code for yourself: https://github.com/sandeepmistry/arduino-nRF5/tree/master/libraries/Wire

There's also issue #176. That might be of some help to you as there has been reported odd behavior on the nRF51 with I2C.

Shreyas-R, 2019-03-18:
Why DMA was not implemented for nRF51 even though it supports as per docs? I see that nRF52 uses TWI manager which works on DMA. Can I use it for nRF51?

I looked myself at the code. Didn't find any reference of interrupt priority or DMA so I posted here.

Don't know whether [they (#176)](https://github.com/sandeepmistry/arduino-nRF5/issues/176) were using softdevice along with I2C or not. Softdevice definitely causes issues as it has highest priority. Nordic suggests to use [Timeslot API](https://devzone.nordicsemi.com/tutorials/b/software-development-kit/posts/setting-up-the-timeslot-api) or use radio notifications to time other low priority works in between. I'm trying to implement that.

dlabun, 2019-03-18:
This project is based off the code base for the Arduino Zero, which does not make use of DMA. A few people have cobbled together DMA support on the Zero but I haven't seen anyone move it to the nRF.

Shreyas-R, 2019-03-19:
Can I use  nRF52 wire library to implement TWIM under nRF51? Or have anybody tried that?

dlabun, 2019-03-19:
You can certainly try at our own risk, I don’t know if there’s any difference between the two libraries.