sandeepmistry/arduino-nRF5 #353: nRF52 ADC External Support

gprotoX, 2019-04-04:
Hello! Thank you for this wonderful support for the board nrf52 ble in Arduino Platform!

I'm reading from the ADC input but I'm reading around 20-30mV from actual value. Default ref is set to 3V right?

If I will use a LM4040 (2.048 / .1%) for external reference will that lower my errors in ADC reading (I had used this before with arduino uno with reading resistances and had accurate results) ? I'm not certain if this is cause by the internal ADC or by the internal REF voltage.

Do you have a plan to add support for external AREF on P0.02 of nrf52?

TIA.

dlabun, 2019-04-04:
The internal AREF is set to 0.6VDC and the external must between 1.0VDC and 2.7VDC. We already have support built in for external AREFs, it's more a question of which board are you using and if it's defined for that board.

gprotoX, 2019-04-04:
I'm using the nrf52 dk and transferred to E73-2G4M04S1B with the same board nrf52 dk selected to follow the pattern of pins from the nrf52dk.

Tried to use analogReference(EXTERNAL) but not working. Went to cores section and found out that wiring_analog for nrf52 has DEFAULT, INTERNAL and VDD/4 options only. I'm currently using the default where it is set to 3.0V. When I set analogReference(EXTERNAL) it will not compile. 

If EXT is supported how should I set it on my code? Thanks.

dlabun, 2019-04-08:
I must have missed the fact that you cannot specify to use the external reference... it’s rather odd because we support specifying which pin to use. I’m going to flag this as a bug for the time being.