sandeepmistry/arduino-nRF5 #354: NRF52832 flash problem

shajek, 2019-04-13:
Hello, i use your core with mysensors library. For flashing is used china ST-link V2 with no problem. I trying some code, flashing , testing, moding, flashing testing - all right. I test one program for 2 weeks, NRF normaly work, but then i want make some changes, plug STlink to USB, upload, and nothing 
`adapter speed: 10000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : clock speed 4000 kHz
Info : STLINK v2 JTAG v33 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.226649
Error: init mode failed (unable to connect to the target)
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 473
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked`

i try switch SWDCLK with SWDIO (for sure), but nothnig... I try made RPI Zero with OpenOCD
`BCM2835 GPIO config: tck = 11, tms = 25, tdi = 10, tdo = 9
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: trst = 7
BCM2835 GPIO config: srst = 18
none separate
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG and SWD modes enabled
Info : clock speed 4061 kHz
in procedure 'init' called at file "openocd.cfg", line 14
in procedure 'ocd_bouncer'`

same problem. i try everything ... connect nrf5 back to PCB my program working, but i cannot reflash it. Can you help me ? Have you expirence with this ? Module is from Ebyte .. thank you

dlabun, 2019-04-13:
You most likely have an electrical issue as programmer is detected by OpenOCD. My only advice is to confirm your connections between the programmer and your board. One common mistake is not powering up the target board.

shajek, 2019-04-14:
Thank you for reply, i already check all connections. I power NRF from programmer and i have permanently soldered SWD cables. I doublecheck all cables with multimeter and measure VCC. Everything is alright (Connected to right pins). As I say, month later I flash smoothly and flawlessly. When i try to connect to target MCU, openocd say "unknown" state ... But when i disconnect one of SWD cable on purpose, Raspberry say:
`BCM2835 GPIO config: tck = 11, tms = 25, tdi = 10, tdo = 9
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: trst = 7
BCM2835 GPIO config: srst = 18
none separate
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG and SWD modes enabled
Info : clock speed 4061 kHz
Info : SWD DPIDR 0x00000001
Error: Could not initialize the debug port`

then i connect cable, and i am back with first error

dlabun, 2019-04-14:
I’m confused, do you have two separate programmers attached at the same time? Most programmers cannot supply enough power for your board to work correctly so I would advise using an external voltage source.

shajek, 2019-04-14:
sorry for confusing :) Nope, i have two separate programmers which i trying separately because i think that st-link just broke. I have one NRF52 with firmly soldered (and checked) cables with VCC GND SWDIO SWDCLK + some i2c and gpio.  When i think that stlink is broken i try connect to MCU with RPi separately with stlink disconnected. Yes i already try external supply with stabilized laboratory power source. That section with disconecting cable is only for purpose of debug if is all connected corectly

dlabun, 2019-04-14:
I am unfamiliar with using a RPi as a programmer so I can suggest getting another ST-Link. We do use a modified version of OpenOCD in this project so attempting to use the RPi may be failing for unrelated reasons.

shajek, 2019-04-14:
Yes, i am waiting for another ST-Link .. but it is very strange... BTW. I make RPI with openocd with this tutorial [https://gitlab.cba.mit.edu/pub/hello-world/nrf52/tree/master](url) and [https://learn.adafruit.com/programming-microcontrollers-using-openocd-on-raspberry-pi/wiring-and-test](url)

both st-link and RPI give me identical error. I will write when i get another stlink or another NRF52 with first stlink

EDIT: i found another NRF52, i connect same ST-Link and RPi and it work.... How can get NRF52 faulty for flash on table,but not faulty for run already flashed program ...

dlabun, 2019-04-14:
It's difficult to say why it happening. It could be anything from something in your code or an external device that's causing it.

shajek, 2019-04-14:
I dont get it, what happening .... I understand what you say, i am litle thinking loud .. I flashed many time same program (little changes in strings), and suddenly ... I use MySensors + BME280 .. it is strange, if i get some clue, i write it here. Thank you for help and your work :)

closed 2019-04-14 by dlabun