sandeepmistry/arduino-nRF5 #358: Direct port access macros not working with secure element ATSHA204

Hoel, 2019-05-03:
I am trying to use the ATSHA204 with the nRF5 arduino core, sadly the direct port access macro are somehow incompatible. I tried all other libraries i could find but they are all based on these macro.

```
/Users/admin/Documents/Arduino/hardware/sandeepmistry/nRF5/cores/nRF5/Arduino.h:96:50: error: invalid conversion from 'NRF_GPIO_Type*' to 'uint8_t {aka unsigned char}' [-fpermissive]
 #define digitalPinToPort(P)        ( &(NRF_GPIO[P]) )
                                                  ^
/Users/admin/Documents/Arduino/libraries/SparkFun_ATSHA204_Library/src/sha204_library.cpp:13:17: note: in expansion of macro 'digitalPinToPort'
  uint8_t port = digitalPinToPort(pin); // temoporarily used to get the next three registers
                 ^
/Users/admin/Documents/Arduino/hardware/sandeepmistry/nRF5/cores/nRF5/Arduino.h:101:44: error: base operand of '->' is not a pointer
 #define portModeRegister(port)     ( &(port->DIRSET) )
                                            ^
/Users/admin/Documents/Arduino/libraries/SparkFun_ATSHA204_Library/src/sha204_library.cpp:16:20: note: in expansion of macro 'portModeRegister'
  device_port_DDR = portModeRegister(port);
                    ^
/Users/admin/Documents/Arduino/hardware/sandeepmistry/nRF5/cores/nRF5/Arduino.h:99:44: error: base operand of '->' is not a pointer
 #define portOutputRegister(port)   ( &(port->OUTSET) )
                                            ^
/Users/admin/Documents/Arduino/libraries/SparkFun_ATSHA204_Library/src/sha204_library.cpp:18:20: note: in expansion of macro 'portOutputRegister'
  device_port_OUT = portOutputRegister(port);
                    ^
/Users/admin/Documents/Arduino/hardware/sandeepmistry/nRF5/cores/nRF5/Arduino.h:100:44: error: base operand of '->' is not a pointer
 #define portInputRegister(port)    ( &(port->IN) )
                                            ^
/Users/admin/Documents/Arduino/libraries/SparkFun_ATSHA204_Library/src/sha204_library.cpp:20:19: note: in expansion of macro 'portInputRegister'
  device_port_IN = portInputRegister(port);
                   ^
exit status 1
```

Does anyone know how it could be fixed?
The library is here https://github.com/sparkfun/SparkFun_ATSHA204_Arduino_Library
 in my case the ATSHA204 is the SWI (wingle wire) version, so it doesnt use Arduino Wire.

dlabun, 2019-05-05:
The nRF chips don't include support for 1-Wire so you're not going to be able to use it. This is a known issue over on the Nordic forums.

closed 2019-05-05 by dlabun

Hoel, 2019-05-06:
Hi, yes that is why the driver implement it with direct port access.
 For now i replaced the statements by API function, but for the sake of efficiency that would be batter if i can use direct register access as it was.