sandeepmistry/arduino-nRF5 #359: Selectable pin for Serial

Spycatcher2k, 2019-05-21:
Would it be possible to add the ability to choose the Serial pins, like I can in uPython? This would make using external serial devices so much easier to interface with.

dlabun, 2019-05-21:
Do you mean selecting which hardware pins are used? That can already be done by changing the board files.

Spycatcher2k, 2019-05-21:
Hi. Thanks for your reply. I was not aware I could do this. I'll have a try
shortly.

Spycatcher2k, 2019-05-21:
Hi - can you give an example of this please - I can not find the section to change!

dlabun, 2019-05-21:
Which board are you using? That way I can make it a little easier to understand.

Spycatcher2k, 2019-05-21:
I'm using the BBC micro:bit. Thanks.

dlabun, 2019-05-21:
So the file you would want to modify is this one:
https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/BBCmicrobit/variant.cpp

If you look down at line 53 and 54 of the file you'll see that currently pin P0.25 is the UART RX and P0.24 is the TX. If you change those two lines you can select another compatible pin for your serial port.

Spycatcher2k, 2019-05-21:
Excellent! Thanks very much.

dlabun, 2019-06-15:
Closing as resolved

closed 2019-06-15 by dlabun