sandeepmistry/arduino-nRF5 #360: std::map Insert Exception

czuvich, 2019-05-22:
I'm trying the std library on the nRF52840 Express. I am able to instantiate the std::vector and std::map just fine. I can also invoke the push_back() method of std::vector; however, when I try to invoke the std::map.insert() method I get the following exception.

var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_547782/sketch/FreeRTOSTimer.cpp.o: In function `std::_Rb_tree_iterator<std::pair<void* const, FreeRTOSTimer*> >::operator--()': /Users/Clay/Library/Arduino15/packages/adafruit/tools/gcc-arm-none-eabi/5_2-2015q4/arm-none-eabi/include/c++/5.2.1/bits/stl_tree.h:220: undefined reference to `std::_Rb_tree_decrement(std::_Rb_tree_node_base*)'

I think it has something to do with the compiler flags. Is there any way I can enable exceptions and std::map.insert() on the nRF52840 Express?

dlabun, 2019-05-22:
Since you're using the Adafruit board please wait for a reply on the Adafruit issue tracker.

closed 2019-05-22 by dlabun

czuvich, 2019-05-23:
The boards from Adafruit are based on the core from here.  A quick question.. I resolved the linker error by replacing the ELF command to use g++ instead of gcc.  Is there a reason why you're using gcc?