sandeepmistry/arduino-nRF5 #361: Black Magic Probe - Cannot access memory at address 0x1c84c

vladkozlov69, 2019-05-31:
Hello,

I'm trying to program NRF52832 using Black Magic Probe made of STM32 (as per https://medium.com/@paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-c013cf2cc38c)

However I'm getting these messages:

> Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52 Access Port
Remote debugging using /dev/cu.usbmodemE0CFB5E71
Attaching to Remote target
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
PC register is not available
PC not available
warning: while parsing target memory map (at line 1): Required element <memory> is missing
Load failed
Cannot access memory at address 0x1c84c
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
PC register is not available
PC not available
Reading symbols from sketch_may17a.ino.elf...done.
Loading section .text, size 0x9dc lma 0x1c000
Starting program: /private/var/folders/3v/mdtlw0dj3tx518cdmcm7q3lw0000gn/T/arduino_build_283132/sketch_may17a.ino.elf 
Program complete!

Am I doing something wrong?

dlabun, 2019-06-15:
Closing due to lack of activity and this doesn’t appear to be an issue with this project.

closed 2019-06-15 by dlabun

pvonmoradi, 2021-01-04:
@vladkozlov69   
1- The blog you referenced used `stlink` platform but you should use `swlink` for the bluepill
2- To flash, try this command:  
```makefile
 	arm-none-eabi-gdb -nx --batch \
        -ex "target extended-remote /dev/ttyACM0" \
        -ex "monitor swdp_scan" \
        -ex "attach 1" \
        -ex "monitor erase_mass" \
        -ex "load" \
        -ex "compare-sections" \
        -ex "kill" \
        $(OUTPUT_DIRECTORY)/nrf52832_xxaa.hex
```