sandeepmistry/arduino-nRF5 #363: Error: couldn't open softdevice hex file

NorHole, 2019-06-11:
Ive been trying to flash the soft device to my NRF52840 DK but its not able to open the hex file. Can someone please help me 

Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Info : No device selected, using first device.
Info : J-Link OB-SAM3U128-V2-NordicSemi compiled Jan 12 2018 16:05:20
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : clock speed 10000 kHz
Info : SWD IDCODE 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Warn : Unknown device (HWID 0x0000015b)
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Error: couldn't open /Users/ThomasA.Anderson/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.1_softdevice.hex
embedded:startup.tcl:454: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 510
at file "embedded:startup.tcl", line 454
Error while flashing SoftDevice.

[image]

pardeepk21, 2019-06-14:
Yeah, the openOCD programmer gives you error of not opening softdevice hex file because it is not present there.

When you install Sandeepmistry's Arduino Core for Nordic NRF5 series, the core does not contain softdevice hex files with them.

You need to download the files separately from Nordic official website and put it in the softdevice folder your openOCD mentioned which in your case is: 

  /Users/ThomasA.Anderson/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/softdevice/s132/hex/

So you need to download softdevice s132 with version 2.0.1 from nordic website.

just download the file and extract it into your nrf5 core folder and you'll be good to go.

hope this helps 

I have attached the file contains softdevice here and also attached the link of Nordic websote, you can download the file from wherever you feel comfortable from:

SoftDevice File:
[s132nrf52201.zip](https://github.com/sandeepmistry/arduino-nRF5/files/3291614/s132nrf52201.zip)

Nordic Website:
https://www.nordicsemi.com/?sc_itemid=%7B10F110DE-001B-4C79-8E5E-1851FC80F751%7D

NorHole, 2019-06-14:
Ok I copied that SoftDevice file and put it in its respective folder and it seemed to work although when I try to upload the blink program it doesn't blink.

This is what came up after I flashed the SoftDevice:

/Users/ThomasA.Anderson/Library/Arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5/bin/openocd -d2 -f interface/jlink.cfg -c transport select swd; set WORKAREASIZE 0; -f target/nrf52.cfg -c init; halt; nrf51 mass_erase; program {{/Users/ThomasA.Anderson/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.1_softdevice.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Info : No device selected, using first device.
Info : J-Link OB-SAM3U128-V2-NordicSemi compiled Jan 12 2018 16:05:20
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : clock speed 10000 kHz
Info : SWD IDCODE 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Error: nrf52.cpu -- clearing lockup after double fault
Polling target nrf52.cpu failed, trying to reexamine
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Warn : Unknown device (HWID 0x0000015b)
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Info : Padding image section 0 with 1692 bytes
Warn : not enough working area available(requested 32)
Warn : no working area available, falling back to slow memory writes
wrote 114688 bytes from file /Users/ThomasA.Anderson/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.1_softdevice.hex in 23.841314s (4.698 KiB/s)
** Programming Finished **
** Verify Started **
Warn : not enough working area available(requested 52)
Warn : not enough working area available(requested 52)
verified 110636 bytes in 1.748919s (61.777 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

This is what came up after I uploaded the Blink program:

Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Info : No device selected, using first device.
Info : J-Link OB-SAM3U128-V2-NordicSemi compiled Jan 12 2018 16:05:20
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : clock speed 10000 kHz
Info : SWD IDCODE 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000548 msp: 0x20010000
** Programming Started **
auto erase enabled
Warn : Unknown device (HWID 0x0000015b)
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 4096 bytes from file /var/folders/pj/nd9t654n6wx4xc_fjgcj6mnr0000gn/T/arduino_build_467765/Blink.ino.hex in 0.212305s (18.841 KiB/s)
** Programming Finished **
** Verify Started **
verified 2744 bytes in 0.049425s (54.217 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

dlabun, 2019-06-15:
The nRF52840 is not supported by this project as it’s not compatible with the S132 v2 soft device, what’s why it won’t program. You’ll need to use one of the supported nRF to use this project.

closed 2019-06-15 by dlabun