sandeepmistry/arduino-nRF5 #368: microbit pin numbering scheme rationale

zerog2k, 2019-07-05:
I see from issue #170 there is confusion surrounding the pin numbering map which arduino-nRF5 uses - I too share the same confusion.

Looking at https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/BBCmicrobit/variant.cpp

It is not clear to me from microbit schematics or any other documentation which I can find why this pin numbering is used.
Can someone explain the rationale behind this arduino pin numbering scheme? I assume it is based on some logic, but I'm trying to determine what it's related to.

Thanks,
Jens

zerog2k, 2019-07-05:
more to the point, is it possible/feasible to implement a "reverse" mapping scheme where we can reference a pin number on the nrf like `P0_13`. Such an example on microbit might be like:
```
#define P0_13 26
```
such that we can do stuff like `digitalWrite(P0_13, HIGH);` ?

dlabun, 2019-07-05:
The pin numbers used in Arduino are a 1 to 1 match for the MCU pin numbers listed in the schematics, the only difference is Arduino drops the P0. prefix.

zerog2k, 2019-07-05:
[image]

that's not at all what I'm seeing...

to give an example

BTN_A is connected to P0.17 (package pin 37)
however in the "arduino pin numbering" scheme here:
https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/BBCmicrobit/variant.cpp#L31
it's considered pin 5

dlabun, 2019-07-05:
I believe you're looking at the schematics for the reference board and not the microbit. The variant is based on these schematics from the BBC Github repository:

https://github.com/bbcmicrobit/hardware/blob/master/SCH_BBC-Microbit_V1.3B.pdf

zerog2k, 2019-07-05:
Ok, on this schematic, BTN_A is still P0.17, but package pin 25.
Still doesnt align with the numbers in variant.cpp

[image]

dlabun, 2019-07-05:
We don't go by the package pin numbers, only the port numbers. If you dump the P0. prefix on the port numbers it does line up.

zerog2k, 2019-07-05:
ok, maybe I wasn't being clear.
I'll give a more concrete example.
In order to read BTN_A, which is connected to P0.17, I would have to do
`digitalRead(5)`

Where does `5` come from? 
Yes it's the index into `g_ADigitalPinMap[]` in variant.ccp, however it's not clear why it is "5" - and not say something logical like `17` for P0.17. I don't see where `5` relates to this in schematics anywhere.  I'm trying to make sense of this pin mapping, and also be able to do logical, intuitive stuff like `digitalRead(P0_17)` or something similar.

dlabun, 2019-07-05:
Ah okay, I understand the question now.... Literally Arduino just starts with what Arduino considers pin 0 and will assign it to the very first hardware port listed in the index and just runs through the list assigning Arduino pin numbers in order. You can change the order if you want based on your needs, the list was set by whoever created the variant with no particular logic that i am aware of.

If you don't like the current naming conventions you can create aliases in the header file, that's already done for the analog pins for example.

sandeepmistry, 2019-07-06:
Please see the pinout diagram here: https://microbit.org/guide/hardware/pins/

closed 2019-07-06 by sandeepmistry