sandeepmistry/arduino-nRF5 #370: I2C and BLEPeripheral doesn't work

kungfupizza, 2019-07-13:
Hi,
I am using this core for nRF52832 and the BLEPeripheral and I2C doesn't work on it, the program gets stuck on blePeripheral.begin() or Wire.begin()

I am using S132 SoftDevice 6.1.0.

Any solutions?

kungfupizza, 2019-07-13:
EDIT: Also tried S132 2.0.1 as said in readme, but still failed.

dlabun, 2019-07-13:
I2C should work fine, which board are you using?

As for BLE issues, please post over on the BLEPeripheral issue tracker as that's a whole separate project and I have zero knowledge of it's inner workings.

kungfupizza, 2019-07-13:
I made a custom board with I2C pull ups of 4.7K on pin 15 and 16, which I changed in the variant.h file.

kungfupizza, 2019-07-13:
Sorry, my mistake. The I2C sensor wasn't getting proper 3.3V and doing some tweaks solved it.

closed 2019-07-13 by kungfupizza