sandeepmistry/arduino-nRF5 #371: Serial does not work very well on the nrf51

thenamelessprogrammer, 2019-07-19:
I have the following serial echo program to test Serial. This is not really a big deal as I have it working,  but I just wanted to point out the issue and maybe get an understanding on why it is happening if anyone else has some ideas. 

```C
/* 
 * Small program to test serial capabilities of the device. This program keeps taking in characters
 * over serial until it receives a newline at which point it echoes back the input to the sending
 * device. To be used along with the python script in this directory for characterization of the
 * serial interface.
 */
/* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *
 * macros
 * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * */
#define STR_LEN 100

/* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *
 * global variables
 * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * */
char myStr[STR_LEN + 1];
bool printLine = false;
int curStrLen = 0; 

/* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *
 * function prototypes
 * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * */
int readLine();

void setup()
{
    /* 
     * Function that is automatically run when the device first boots up used for initializing the
     * device.
     */
    Serial.begin(115200);
    for(int i = 0; i <= STR_LEN; i++)
    {
        myStr[i] = '\0';
    }
}

void loop() 
{
    /*
     * Main program loop. This function is called after the setup function and loops forever.
     */
    if (Serial.available() > 0)
    {
        readLine();
        if(printLine)
        {
            Serial.println(myStr);
            printLine = false;
            curStrLen = 0;
        }
    }
}

int readLine()
{
    /*
     * Function that is called whenever there is serial data available. It reads the current data
     * available on the serial line and stores it in the array that contains the string. If the end
     * of the string is reached indicated by the reception of a newline character is sets the
     * printLine to true so the line can be printed.
     * Global variable modifications:
     *      - printLine: set to true if end of line is reached.
     */
    Serial.print("in rl curStrLen = ");
    Serial.println(curStrLen);
    char incomingByte;

    Serial.println(Serial.available());
    while(Serial.available() > 0 && curStrLen < STR_LEN)
    {
        incomingByte = Serial.read();
        if(incomingByte != '\r' && incomingByte != '\n')
        {
            myStr[curStrLen++] = incomingByte;
        }
        else if (incomingByte == '\n')
        {
            printLine = true;
            break;
        }
    }
    myStr[curStrLen] = '\0';
    Serial.print(Serial.available());
    Serial.println(" = Serial.available()");
    Serial.print("Leaving rl curStrLen = ");
    Serial.println(curStrLen);
    return curStrLen;
}
```

When I upload this code to the feather M0 and send the string "This is a test" over Serial with a carriage return and a newline. It correctly shows that 16 is the number of available characters, and reads all the characters with one function call.

```
23:14:32.620 -> 16
23:14:32.620 -> 0
23:14:32.620 -> Leaving rl curStrLen = 14
23:14:32.620 -> This is a test
```

When I upload the same code to my nrf51 DK (PCA10028) and send the string "This is a test" over Serial with a carriage return and a newline. It first shows that 1 is the number of available characters. It then reads 4 characters and returns to the main loop. The function then gets called again and the rest of the characters are read.

```
23:12:22.102 -> 1
23:12:22.102 -> 0
23:12:22.102 -> Leaving rl curStrLen = 4
23:12:22.102 -> 12
23:12:22.102 -> 0
23:12:22.102 -> Leaving rl curStrLen = 14
23:12:22.102 -> This is a test
```

Does anyone know why this discrepancy exists between the M0 and the nrf51? 

I first noticed the issue when I was trying to write a readline function with a local variable that keeps track of the number of letters in the string, but then after I realized what the issue on the nrf51 was I made it a global variable to get around it.

jandrassy, 2019-07-19:
it is simple, on nRF51 you can read faster then the buffer fills with transported bytes. 
is it native USB on M0? what USB chip do you use for the nRF51?

thenamelessprogrammer, 2019-07-19:
I know that it reads 1 byte at a time. The number Serial.available() returns is the number of bytes that are available to be read off the buffer. I  also know that on the nrf51 it is reading faster then the UART fills the buffer with the transported bytes.  My question was why is that the case? I guess I got the answer though based on the info below:

I am using the nRF5x DK with SEGGER J-Link OB chip. I read that this has a has emulated COM port which works like serial link. So I am assuming that why it is filling up slower than the feather M0. I am not sure what the feather m0 has though.

jandrassy, 2019-07-19:
Feather M0 has native USB of the SAMD21G MCU

closed 2019-07-22 by thenamelessprogrammer