sandeepmistry/arduino-nRF5 #375: Looking for a working hex flashing method in windows 10

gmfcd128, 2019-08-09:
I am currently on Windows 10 1809 and is unable to replace my J-Link driver using Zadig successfully.
I grabbed the generated hex from C:\Users\(My Username)\AppData\Local\Temp, loaded it into nrfgo studio, erased the device and flashed it, Bluetooth was not working.
When I flashed softdevice s132 2.0.2 using nrfgo studio, the main program seemed to be wiped during the process. Flashing the generated hex file afterward produced the opposite result.
Does anyone have an idea on how to make this work?

jandrassy, 2019-08-09:
the build result of your sketch has fixed addresses. if you build without soft-device, the sketch starts at address 0x0. with soft-device the start address is set according to the size of the soft-device

gmfcd128, 2019-08-09:
@jandrassy I selected softdevice s132 in Arduino IDE before compiling the firmware.

dlabun, 2019-10-03:
Closing due to lack of activity

closed 2019-10-03 by dlabun