sandeepmistry/arduino-nRF5 #376: softDevice can't get past download error

aguileraGit, 2019-08-24:
I have followed the instructions on the ReadMe as well as #312 . 

I have downloaded the softdevice and placed them in the directory.
```
Diegos-MacBook-Pro-709:hex diegoaguilera$ pwd
/Users/diegoaguilera/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/softdevice/s130/hex
Diegos-MacBook-Pro-709:hex diegoaguilera$ ls
s130_nrf51_2.0.1_softdevice.hex
```

Select the device in Arduino.
[image]

And the error message.
```
Arduino: 1.8.9 (Mac OS X), Board: "Generic nRF51, 16 kB RAM, 256 kB flash (xxaa), S130, Crystal Oscillator"

Downloading 'http://www.nordicsemi.com/eng/content/download/95150/1606929/file/s130_nrf51_2.0.1.zip' ...
Error while flashing SoftDevice.
java.io.FileNotFoundException: http://www.nordicsemi.com/eng/content/download/95150/1606929/file/s130_nrf51_2.0.1.zip
```

aguileraGit, 2019-08-24:
Flashed it manually using the [JLink Programmer and these instructions](https://devzone.nordicsemi.com/f/nordic-q-a/9695/flash-softdevice-and-then-an-app-using-jlink). Then selected softdevice in Arduino and programmed like normal. 

I just tried the iBeacon code and I see it. Guess it's working?!

closed 2019-08-29 by aguileraGit

pixmix, 2019-09-23:
I am not sure this is helpful, especially considering that nordic seems to change often their download locations.
I have edited the file softdevices.txt in /home/pm/Arduino/hardware/sandeepmistry/nRF5 on Ububtu platform and it works fine today.
I attach here my version of the file for reference.
[softdevices.txt](https://github.com/sandeepmistry/arduino-nRF5/files/3640521/softdevices.txt)