sandeepmistry/arduino-nRF5 #380: Serial is not working on generic nrf52832 board

mayurharge, 2019-09-20:
I am using nrf52832 board 
uploaded basic sketch (blink without delay & Serial print combined)
```
 Serial.println("buttonState");
  delay(1);
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }
    digitalWrite(ledPin, ledState);
  }
```
Now the blink is working but serial is not
Other details:
1. Upload method j-lnk
2. baudrate (9600)
3. using FTDi connector for serial communication

SCH:- SparkFun board
LSE crystal 32khz
HSE crystal 16Mhz

dlabun, 2019-09-23:
Which Sparkfun board are you using?

mayurharge, 2019-09-24:
https://www.sparkfun.com/products/13990

We are using this board.

tcpipchip, 2019-09-24:
Did you make the changes on VARIANTS files ? (TXD and RXD pins)

mayurharge, 2019-09-24:
No

tcpipchip, 2019-09-24:
variants.h e variantes.cpp must to be modified to GPIO used by sparkfun. Maybe is redirecting to other port the UART

dlabun, 2019-10-03:
Closing due to lack of activity

closed 2019-10-03 by dlabun