sandeepmistry/arduino-nRF5 #383: Problem using nRF5-DK to program nRF51822 povered at 1.8V via logic level translator

pixmix, 2019-10-07:
I connected my prototype to nRF5-DK (PCA10028) port 19 as described in the relevant documentation via a standard logic level translator based on BSS138 mosfet.
I kept getting for a couple of days errors of the type:

cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
in procedure 'program'
in procedure 'init' called at file "embedded:startup.tcl", line 473
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

An error occurred while uploading the sketch

Then I checked with oscilloscope to discover that the pulse shape through the logic level translator was degraded.
I spent some time searching resources on the network trying to figure out how to change the speed of communication from 1000 kHz to something more suitable.
Finally I managed to get the thing working changing two files (one might have been enough but I prefer to follow the architecture of the configuration and removed the detail from the general case to specify it in the detail case).
What I changed is:
a) file $(HOME)/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5/scripts/target/nrf51.cfg:Line52:
# adapter_khz 1000 (commented this setting)
b) file $(HOME)/Arduino/hardware/sandeepmistry/nRF5/boards.txt
{specific board}.upload.setup_command=transport select swd; set WORKAREASIZE 0; adapter_khz 330 (added the command adapter_khz 330 to the existing OCD command line)

I did not test the impact of this change on other boards, but I expect that having removed the setting from the first file I may need to specify it in the boards.txt file for every relevant board.

I report this here because I imagine others may face the same issue and I was not able to find a better way to solve it through the documentation and forums.
If there is a better way to set the upload speed of the j-link OpenOCD SWD interface I'd love to know that. If not then I hope that this may help the maintainer including the change in a proper way should it be considered an interesting feature.

dlabun, 2019-10-07:
What kind of programmer are you using? There should be no need for a level translator unless you are using a 5V Arm processor and they're quite rare.

pixmix, 2019-10-08:
Thanks @dlabun , 
I have a Nordic nRF51-DK (PCA10028) that only supports 3.0V, while my custom nRF51822 based board is powered 1.8V.
As explained in the documentation I need the logic level translator.

dlabun, 2019-10-08:
I highly recommend using a programmer designed for 1.8v operation, otherwise you’re going to have issues. Both the J link and BMP have designed in circuits that can shift down to 1.8v and are known to work. Outside of that, modifying the programmer file like you did is your only option as you’re not using what I would consider a normal setup to program your board.

closed 2019-10-08 by dlabun