sandeepmistry/arduino-nRF5 #384: microbit reset button ignored on battery

Cat30Mk3, 2019-10-16:
This issue has been reported and patched with bbcmicrobit/micropython and exists using  this arduino-nRF5 core.

See https://github.com/bbcmicrobit/micropython/pull/550.

dlabun, 2019-10-16:
I do not believe there's any issue on our end as that button labeled reset isn't wired into the reset circuit of the nRF. If you're using micropython you're just doing a soft reset from that button.

Cat30Mk3, 2019-10-17:
Here's what the microbit/micropython developer said about this issue:
I believe he built a patch to handle it behind the scenes. Sorry if I'm being dense about this and do set me straight if I'm m off track,  but could you not do the same? 

""Thanks for the report. Indeed we must configure the soft reset button manually at start up. See #550 for a patch that does this.

Note that there is a slight difference with reset behaviour when running on battery: the reset occurs as soon as you press the button, whereas when running on USB holding the reset button down will hold the device in reset until you let go of the reset button. This may have real consequences for users, eg they want to start a timer upon reset and hold down the reset button until they are ready to start timing. To have the same behaviour on battery the start-up code could check if reset is held down and wait until it is released before it starts executing. The downside of doing this is that it's different to how the C++ code would work (and how makecode works)."

dlabun, 2019-10-17:
This is an apples vs oranges situation. Microbbc/Micropython has an interpreter running on the nRF chip that takes care of things like watching for a soft reset button push and executing the code your wrote. The interpreter will initiate a reset even if your code is totally broken. What that patch was fixing was a bug in the Micropython interpreter were the soft reset feature didn't work as expected.

On the other hand, we have no such interpreter to handle such tasks. To reset the board it's assumed that you are going to do it via the nRF's hardware reset circuit or from your own code. If you wanted to include this feature you'd have to write into the code you write (or if you're really savvy, a new Arduino library specific for the Microbit.)

Cat30Mk3, 2019-10-18:
Ok, well thanks for the explanation. I’m sure there will be others bumping into this. I’m not sure how to tackle this on my own but I will give it a try. Cheers.

closed 2019-10-18 by Cat30Mk3