sandeepmistry/arduino-nRF5 #388: The nRF52840DK not compatible with nRF52DK Variant

chappy1978, 2019-11-21:
I have been working with a nRF52840DK for a project and when I use platformio.ini below
[env:nrf52840_dk]
platform = nordicnrf52
board = nrf52840_dk
framework = arduino

I get the wrong pins outputs from the Dev board.  With a simple blinky.cpp that runs through the pins causes the wrong pins to light up. Also many of the pins dont seem to be mapped to Arduino framework at all. Its seems like the nRF52840DK needs its own variant files not reusing the variants from a different board. I have included a pinout from Nordic for the nRF52840DK
[image]
 

#include <Arduino.h>
int led = 0;

void setup() {
  Serial.begin(115200);
}

// the loop routine runs over and over again forever:
void loop() {
  pinMode(led, OUTPUT);
  if(led >= 10){led = 0;}
  Serial.println(led);

  digitalWrite(led, HIGH);   
  delay(500);              
  digitalWrite(led, LOW);    
  delay(500);               
  led++;
}

dlabun, 2019-11-21:
This is a PlatformIO issue as this project doesn't include support for the nRF52840DK, please contact them for help.

chappy1978, 2019-11-21:
Why? I just wrote a new variant.h and .cpp for your project and its working. 
[Framework bug.zip](https://github.com/sandeepmistry/arduino-nRF5/files/3874883/Framework.bug.zip)

dlabun, 2019-11-21:
My apologies, I took this as you were asking us to fix PlatformIO. If you want to submit a variant for the nRF52840DK I'd gladly take it.