sandeepmistry/arduino-nRF5 #389: Wire bug when requesting more than 64 bytes via Wire.requestFrom

mpava, 2019-11-22:
Hi I think there is a bug in your Wire implementation.

nRF5/RingBuffer.h defines SERIAL_BUFFER_SIZE 64

This makes it impossible to fetch more than 64 bytes when calling:

Wire.requestFrom(address, quantity)

When requesting 64 bytes and more, you get odd behaviours, such as infinite looping or the ring buffer, runaway memory override causing a crash.

dlabun, 2019-11-23:
I wouldn’t call this a bug but rather just a limitation of the library? What’s the use case that you need to move that much data in a single transaction?

mpava, 2019-11-25:
I am reading/writing to/from an EEPROM over I2C. The use-case is to read a page of 128/256 bytes from the memory. As a dirty fix I locally extended the SERIAL_BUFFER_SIZE to 257 bytes, which works for me for now. This is not very memory efficient for normal use-cases, when you read just a couple of bytes from sensors etc. I guess a proper fix would be to have some sort of dynamic memory allocation based on returned data size, with some limit and error if the limit is breached. :)

dlabun, 2019-11-27:
I'm going to close this as a Won't Fix since you have a local work around and this isn't a major issue with the project. If you come up with a permanent fix feel free to submit a PR for it.

closed 2019-11-27 by dlabun