sandeepmistry/arduino-nRF5 #393: nRF52840 module (E73) pinout reference for Arduino IDE pins?

mrdc, 2019-11-28:
Hello,

I'm trying to play with nRF52840 module by EByte and can't find or understand how Arduino IDE handles it's pin assigment to a module/microprocessor pin.
[image]
[image]
Pins:
[image]
[image]
[image]
[image]
How to find correct diagram which shows which nRF52840 pin corresponds to Arduino IDE pin?
I have it only for 52832 module:
[image]

dlabun, 2019-11-28:
There’s no variant for this board so you’d need to make one or use the generic. You can find more information about variants and 3rd party boards on the Arduino website.

mrdc, 2019-11-28:
>  or use the generic.

What should I do if use generic nRF52? I haven't find any info related to pin assigment.

mrdc, 2019-11-28:
For example for Adafruit 52840 pins are defined:

```c
const uint32_t g_ADigitalPinMap[] =
{
  // D0 .. D13
  25,  // D0  is P0.25 (UART TX)
  24,  // D1  is P0.24 (UART RX 
  10,  // D2  is P0.10 (NFC2)
  47,  // D3  is P1.15 (LED1)
  42,  // D4  is P1.10 (LED2)
  40,  // D5  is P1.08
   7,  // D6  is P0.07
  34,  // D7  is P1.02 (Button)
  16,  // D8  is P0.16 (NeoPixel)
  26,  // D9  is P0.26
  27,  // D10 is P0.27
   6,  // D11 is P0.06
   8,  // D12 is P0.08
  41,  // D13 is P1.09

  // D14 .. D21 (aka A0 .. A7)
   4,  // D14 is P0.04 (A0)
   5,  // D15 is P0.05 (A1)
  30,  // D16 is P0.30 (A2)
  28,  // D17 is P0.28 (A3)
   2,  // D18 is P0.02 (A4)
   3,  // D19 is P0.03 (A5)
  29,  // D20 is P0.29 (A6, Battery)
  31,  // D21 is P0.31 (A7, ARef)

  // D22 .. D23 (aka I2C pins)
  12,  // D22 is P0.12 (SDA)
  11,  // D23 is P0.11 (SCL)

  // D24 .. D26 (aka SPI pins)
  15,  // D24 is P0.15 (SPI MISO)
  13,  // D25 is P0.13 (SPI MOSI)
  14,  // D26 is P0.14 (SPI SCK )

  // QSPI pins (not exposed via any header / test point)
  19,  // D27 is P0.19 (QSPI CLK)
  20,  // D28 is P0.20 (QSPI CS)
  17,  // D29 is P0.17 (QSPI Data 0)
  22,  // D30 is P0.22 (QSPI Data 1)
  23,  // D31 is P0.23 (QSPI Data 2)
  21,  // D32 is P0.21 (QSPI Data 3) 
 ```

dlabun, 2019-11-28:
It's mapped one to one, Pin 1 is P0.01, Pin 2 is P0.02, etc.

mrdc, 2019-11-28:
Thanks! And if it's pin P1.00, P1.02, P1.09 or P12, P15, P17? How will be defined in Arduino IDE?

closed 2019-12-05 by mrdc

Yousefff1, 2021-11-20:
@mrdc have you managed to get this to work with the Arduino IDE?

corteXnzt, 2022-03-18:
@Yousefff1 

I know this is late but in case you still need, I have been using e73-2g4m04s1b. When compiling I just select nRF52-dk (nrf52832) and adapt the numbering of pins of DK to the e73-2g4m04s1b.

Borkuz, 2023-11-15:
> @Yousefff1 
> 
> I know this is late but in case you still need, I have been using e73-2g4m04s1b. When compiling I just select nRF52-dk (nrf52832) and adapt the numbering of pins of DK to the e73-2g4m04s1b.

Do you mind sharing it anyways

Borkuz, 2023-11-15:
> @Yousefff1 
> 
> I know this is late but in case you still need, I have been using e73-2g4m04s1b. When compiling I just select nRF52-dk (nrf52832) and adapt the numbering of pins of DK to the e73-2g4m04s1b.

Do you mind sharing it anyways?

corteXnzt, 2023-11-15:
> > @Yousefff1
> > I know this is late but in case you still need, I have been using e73-2g4m04s1b. When compiling I just select nRF52-dk (nrf52832) and adapt the numbering of pins of DK to the e73-2g4m04s1b.
> 
> Do you mind sharing it anyways?

What would I share? just follow the pins indicated in nrf52dk board and check the datasheet of e73-2g4m04s1b for same numbering.

Borkuz, 2023-11-15:
Cool I will give it a try, kinda new to creating these pin out things. Thanks for leading the way!