sandeepmistry/arduino-nRF5 #394: ST-Link v2.1 support?

mrdc, 2019-11-29:
Hello,

I'm using ST-Link V2 to flash nRF52 chips - works fine. Then I've decided to try newer ST-Link V2.1:
I've changed the original V2 vid_pid `hla_vid_pid 0x0483 0x3748` in` \sandeepmistry\tools\openocd\0.10.0-dev.nrf5\scripts\interface\stlink-v2.cfg` to my ST-Link V2.1 vid_pid: `hla_vid_pid 0x0483 0x374a`. 
But flashing is not working - I receive this error:

> the selected serial port 
>  does not exist or your board is not connected

Full log:
<details>
```
C:\Users\111111111\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/stlink-v2.cfg -c transport select hla_swd; set WORKAREASIZE 0x4000; -f target/nrf52.cfg -c program {{C:\Users\111111111\AppData\Local\Temp\arduino_build_947131/xxxxxx.ino.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 10000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : clock speed 4000 kHz
Error: read version failed
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 473
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

the selected serial port 
 does not exist or your board is not connected
```
</details>
ST-Link V2.1 is recognized by ST-Link Utility, so it's 100% working.

Any ideas how to fix?

dlabun, 2019-11-29:
The ST-Link v2.1 is not supported as it's a completely different programmer. I'd recommend you stick with the v2.0.

mrdc, 2019-11-30:
Asking about V2.1 mainly because it has Virtual Com port - need it to use Serial output.
Will J-Link OB V7 (V9) work with Serial output for nRF52 chips?

mrdc, 2019-12-05:
> Will J-Link OB V7 (V9) work with `Serial.println("")` for nRF52 chips?

Unswering my own question - only 7.2+.

closed 2019-12-05 by mrdc