sandeepmistry/arduino-nRF5 #395: Support for nRF52840

mpava, 2019-12-18:
Hi Sandeep

I am exploring adding support for nRF52840. I checked Nordic's website, it requires new softdevice S140. I looked through your nRF5 packages and it seems that it may not be too hard.  It seems this will require some tweaks in BLEPeripheral/src/nRF51822.cpp as the API has changed a bit in S140.

What do you think will be the gotchas?

I will need to create an s140/toolchain/armgcc/armgcc_s140_nrf52840_xxaa.ld file. Where did you get the ORIGIN values from for the S312? I can see the size of FLASH and RAM is documented in the Nordic's docs that come in the S140 ZIP softdevice ZIP file. They don't mention the origin values though.

Please give us some advice. I will aim to contribute this back when I get it working.

thanks

Mike

dlabun, 2019-12-18:
The biggest gotcha is going to the be difference in Nordic SDK versions between what we're using and what the 840 supports. At some point the SDK drops support for the nRF51 and then a few versions later adds the nRF52840. If you can find a way to keep supporting the old nRF51 and add the new 840 I'll gladly accept it. There may be a licensing land mind as well, Nordic added some questionable language that made me second guess the ability to redistribute their SDK files.

As for the ORGIN values, I don't have a clue were Sandeep got those.... May be it arbitrary?

lefedor, 2020-05-06:
Hello, @mpava, have You got any workable result with 52840 ?

bhelm, 2020-05-22:
I also would like to see it supported

kiafaldorius, 2020-06-14:
I know I'm late to this, but as previously mentioned, the nRF52840 requires an entirely different softdevice, S140, for long range in addition to the SDK dropping support of nRF51. The SDK that has support for the 840 came after nRF51 support was dropped. This makes backward compatibility difficult to achieve without packaging a separate subsystem specifically for the 840.

Although, there is the Adafruit nRF52 Arduino project that specifically targets the 52840, so forking that project and adding support for your board will be a much easier endeavor than upgrading this project. https://github.com/adafruit/Adafruit_nRF52_Arduino

Good luck!

IoTThinks, 2020-06-14:
For nRF, can try RadioLib.
They are totally different radio chip.

lefedor, 2020-06-14:
> Although, there is the Adafruit nRF52 Arduino project that specifically targets the 52840, so forking that project and adding support for your board will be a much easier endeavor than upgrading this project. https://github.com/adafruit/Adafruit_nRF52_Arduino
> 
> Good luck!

This is really flexible platform still I have better ultra-low-power results with non-rtos environment for nrf32. This is why it's interesting to have capability use this one for nrf40.

taha-embi, 2026-01-02:
@mpava were you successful in adding support for nrf52840 for soft device s140 or not??