sandeepmistry/arduino-nRF5 #399: sketches not starting on bare-bone MCU

lyakh, 2020-01-19:
Hi, I'm using a bare-bone E73-2G4M04S-52832 module and if I use OpenOCD to flash the SparkFun bootloader to it and then use SparkFun nRF5283 Arduino libraries, I can build and run sketches on it. But then I tried running without SparkFun code, using a slightly modified version of your image from https://www.instructables.com/id/Very-Low-Power-BLE-Made-Easy-With-Arduino/ and sketches don't run.
What I see is, that at the softdevice code is at 0x0, where it has its vector table. At 0x4 is the reset vector, which is taken upon reset - I can see that in gdb. The sketch builds to be loaded at 0x1c00,0 and it also has a vector table in the beginning. But its reset handler or its main() function are never entered. Is this how it is supposed to be? Are sketches supposed to be built for 0x1c000 and should they have a vector table in the beginning? How should a sketch be invoked - is the softdevice code supposed to switch over to a vector table at 0x1c00 and issue a reset, or should it just jump to the reset handler itself, or what's the expected behaviour? Is this actually softdevice code that does that or is it your support code?

ab3lson, 2021-03-31:
Did you ever get this working? I am having the same issue

drmpf, 2022-02-25:
I am the author of that very low power instructable.
My first suggestion is to try an un-modifed version of the Sandeedmisty's package.  Install V0.7.0 and see if you can get that running for your module loading the softdevice and then a simple blink sketch
Once that is running move onto the latest version of very-low-power  
[Easy Very Low Power BLE in Arduino --2022](https://www.forward.com.au/pfod/BLE/LowPower_2022/index.html)
Drop my a line via pfod.com.au at support with how you get on.