sandeepmistry/arduino-nRF5 #406: Core development can build just fine, but not able to find openocd

ekawahyu, 2020-02-28:
My platform is Ubuntu 18.04.3 LTS. Arduino IDE 1.8.10. For Core Development, I followed the given instruction to clone this repo in `~/Arduino/hardware/sandeepmistry/nRF5`. I copied the `tools` folder when installing through Board Manager and put it in `~/Arduino/hardware/sandeepmistry/tools`. After all of Core Development setup was done, I removed the original nRF5 installation through Board Manager, and closed Arduino IDE.

The next time I launched Arduino IDE, I could build the Blinky project just fine, but it failed to find the OpenOCD. Here is the error message I got:

```
/home/ubuntu/Arduino/hardware/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-gcc -L/tmp/arduino_build_281779 -Os -Wl,--gc-sections -save-temps -L/home/ubuntu/Arduino/hardware/sandeepmistry/nRF5/cores/nRF5/SDK/components/toolchain/gcc/ -L/home/ubuntu/Arduino/hardware/sandeepmistry/nRF5/cores/nRF5/SDK/components/softdevice/none/toolchain/armgcc/ -Tnrf52_xxaa.ld -Wl,-Map,/tmp/arduino_build_281779/Blink.ino.map --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o /tmp/arduino_build_281779/Blink.ino.elf /tmp/arduino_build_281779/sketch/Blink.ino.cpp.o /tmp/arduino_build_281779/core/variant.cpp.o -Wl,--start-group -lm /tmp/arduino_build_281779/../arduino_cache_561663/core/core_sandeepmistry_nRF5_nRF52DK_softdevice_none_4a4607ee21ecf7cb26ae72b79d8bd27a.a -Wl,--end-group
/home/ubuntu/Arduino/hardware/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-objcopy -O binary /tmp/arduino_build_281779/Blink.ino.elf /tmp/arduino_build_281779/Blink.ino.bin
/home/ubuntu/Arduino/hardware/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-objcopy -O ihex /tmp/arduino_build_281779/Blink.ino.elf /tmp/arduino_build_281779/Blink.ino.hex
/home/ubuntu/Arduino/hardware/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-size -A /tmp/arduino_build_281779/Blink.ino.elf
Sketch uses 2628 bytes (0%) of program storage space. Maximum is 524288 bytes.
{runtime.tools.openocd-0.10.0-dev.nrf5.path}/bin/openocd -d2 -f interface/jlink.cfg -c transport select swd; -f target/nrf52.cfg -c program {{/tmp/arduino_build_281779/Blink.ino.hex}} verify reset; shutdown; 
java.io.IOException: Cannot run program "{runtime.tools.openocd-0.10.0-dev.nrf5.path}/bin/openocd": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

My temporary workaround is modifying the path to OpenOCD manually, pointing the absolute path of it as you can see below:

```
#tools.openocd.path={runtime.tools.openocd-0.10.0-dev.nrf5.path}
tools.openocd.path=/home/ubuntu/Arduino/hardware/sandeepmistry/tools/openocd/0.10.0-dev.nrf5
```

Anyone knows how to fix this issue? My understanding is that, from the way the `tools.openocd.path` is defined, `~/Arduino/hardware/sandeepmistry/tools` is the right place to look for OpenOCD. Please correct me if I am wrong. Thank you.

sandeepmistry, 2020-12-02:
@ekawahyu can you try cloning into `~/Arduino/hardware/sandeepmistry-github/nRF5`?

Maybe the read me instructions are not compatible with newer IDE releases.

ekawahyu, 2020-12-02:
@sandeepmistry what Arduino version the instruction was written of? I have tried several old versions with no luck either.

I think there might be a bug in the Arduino IDE, but I just don't know how to debug Java. My understanding is the same as yours, when the default package installation is removed, the IDE should look into the hardware folder.

Detail finding was reported here: https://github.com/arduino/Arduino/issues/10761, but no follow up so far and consider as invalid.

So, in summary, there is a workaround I made if I have to develop under Arduino/hardware. The {runtime.tools.<vendor_tool>.path} can be temporarily replaced with {runtime.hardware.path} plus the absolute path down to the upload script. But again, this is troublesome because everytime I need to push/merge my changes, I need to remind myself to put it back to {runtime.tools}

Personally, I don't like to develop under `.arduino15` because I may accidentally remove it from the IDE. I hope you can help to find and solve this issue.

ekawahyu, 2020-12-02:
By the way, I mainly on Ubuntu 18.04 and 20.04. This is not platform specific because I see the same behavior or macOS and Windows PC as well. Same workaround works for all platform. This is not target specific, because I also see the same behavior for STM32Duino.

sandeepmistry, 2020-12-03:
@ekawahyu I would not recommend moving the tools folder to the sketchbook dir.

My flow is to install the nRF5 core via the boards manager (IDE 1.8.13), and then:
```
git clone https://github.com/sandeepmistry/arduino-nRF5.git <sketchbook>/hardware/sandeepmistry-github/nRF5
```
(I'll update the read me now to add `-github`)

You can then see two nRF5 entries when selecting a board, one will have "in sketchbook" in the title:

[image]

ekawahyu, 2020-12-03:
Ok, I think I didn't try this one because I was afraid of getting board duplicates. Good to know. Let me know if you need to test anything after you done with instruction update.

This may solve the issue with code development, but it would also be great to have tools folder working under sketchbook (in case anyone would like to test latest tools, upload script, or whatever).

sandeepmistry, 2020-12-03:
> Ok, I think I didn't try this one because I was afraid of getting board duplicates. Good to know. Let me know if you need to test anything after you done with instruction update.

Instructions have been updated on the read me, feel free to test them out.

> This may solve the issue with code development, but it would also be great to have tools folder working under sketchbook (in case anyone would like to test latest tools, upload script, or whatever).

The latter would be out of scope for the goals of this project.

ekawahyu, 2020-12-03:
Alright, I am closing this issue. Core development in sketchbook location can now use the tools installed by board manager.

There seems to be an issue with SoftDevice flashing, but that's another ticket that I am gonna open soon.

closed 2020-12-03 by ekawahyu

ekawahyu, 2020-12-03:
@sandeepmistry I have been investigating further about SoftDevice flashing issue. It happens only with the core located in the sketchbook. Flashing SoftDevice can be done once, right after cloning the repo into the hardware folder.

The bug is reproducible, here are the steps where you can successfully flash SoftDevice:
1. Clone the repo in hardware folder in sandeepmistry-github/nRF5
2. Open Arduino IDE
3. Select Nordic Semiconductor nRF52DK (in sketchbook)
4. Select SoftDevice S132
5. Select serial port /dev/ttyACM0
6. This is important, **DO NOT SELECT** any programmer, none of the programmer should be selected (weird???)
7. Select nRF5 Flash SoftDevice

After SoftDevice flashing device is done, everything should work just fine. In order to upload the binary, the programmer needs to be selected and this is where the problem begins. Once a programmer is selected, if you need to reflash SoftDevice, Arduino IDE gives you `Unsupported programmer!`

Feel free to re-open this issue if you think it is worth fixing, otherwise my current workaround is by using the default board manager installation to flash the SoftDevice.