sandeepmistry/arduino-nRF5 #409: Error while flashing SoftDevice. when flashing RAK itracker v3 board in s132

alexjolly28, 2020-03-16:
Downloading 'http://www.nordicsemi.com/eng/content/download/95151/1606944/file/s132_nrf52_2.0.0.zip' ...
.done
Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
in procedure 'init' 
in procedure 'ocd_bouncer'

Error while flashing SoftDevice.

Entretoize, 2020-11-28:
Same error for me, did you find a solution ?

drmpf, 2022-02-25:
Looks like softdevice s132 was not downloaded.
V0.7.0 worked for me, try dumping what you have and install V0.7.0
If you module is 'protected' then you need to un-protect it as covered here
https://www.forward.com.au/pfod/BLE/LowPower_2022/index.html#remove