sandeepmistry/arduino-nRF5 #413: DWM1001C variant

pmondejar97, 2020-05-15:
I would like to use the DWM1001C but the variant that is already developed do not match with this one. How can the variant be adapted to this module?

markshancock, 2020-06-30:
I tried it too.  The installation instructions seem straight-forward; but, I have a DWM1001-DEV which has a special programming interface (JLink-Flash) and I couldn't figure out how to make that work.  We have a real JLink; but, it is at the office under COVID quarantine.

jenguidanos, 2020-07-02:
Perhaps I can contribute more information to @pmondejar97's original question (we are co-workers). 

This is the documentation of the PCB developed by Pablo, based on the nRF52832 (a work in progress):
https://rhomb.io/docs/master-modules/dwm1001c/documentation/

Probably the main question is which variant to use as a basis for developing our own. The logical answer I think would be to use the variant [DWM1001-DEV](https://github.com/sandeepmistry/arduino-nRF5/tree/master/variants/DWM1001-DEV). I have doubts about the IRQ pins, which I don't see in Pablo's documentation or they appear under a different name.

In short, we would like to know if the procedure would be as follows:
* Clone the repository
* Setup Arduino IDE following the instructions in the readme.md file
* Create a new variant folder in our local repo based on DWM1001-DEV
* Edit the variant.h/cpp files to route the pins according to our design
* Use the J-Link interface for flashing and debugging

Besides creating the variants directory, is there any other file that needs to be modified for Arduino IDE to recognize our PCB?