sandeepmistry/arduino-nRF5 #416: Any app timer support for nRF51822

RuhanSA079, 2020-07-12:
Hello, I am developing a device that will run for a long time in sleep mode.

I realised that I cannot use RTC1(Arduino millis, micros) or RTC0 (Softdevice uses RTC0) to generate a interrupt after going to sleep. (sd_app_evt_wait)

Is it possible to use the app timer, as specified by the Nordic SDS?
I did not see any app timer related files in the SDK in Arduino IDE.

Erickrk, 2020-10-06:
Hey, @RuhanSA079 have you seen this issue [https://github.com/sandeepmistry/arduino-nRF5/issues/347](url) ? I was searching for a way to use Sleep mode with this core and ran into it, gonna start look more into the info he shares now.

RuhanSA079, 2020-10-06:
I tried to modify the core files to include some sort of deep sleep mode, but ended up using some test code from atc1441 ( https://github.com/atc1441/ATCwatch/tree/master/ATCwatch ) to achieve the low power sleep mode. The only problem I now have, is after a while 8mins, the RTC counter overflows, and waking up the device. I resolved this by go going back to sleep immediately. (simple state machine checking)
Another problem I have, the RTC's counter (millis()/micros() on the Arduino core is being used by Paul Stoffregen's Time.h lib) which results the time getting out by days and months after a day or so in this sleep mode I implemented. I suspect the RTC overflowing stuffs is not handled correctly after a while?
I don't know how to fix this.

RuhanSA079, 2020-10-06:
Seems like my attempts at getting app timer support built in my code failed horribly (needs too many deps) for just one app timer.

Erickrk, 2020-10-06:
Thanks for the update! 
The problem related to the counter's overflow you described looks like the same one as the one pointed out in #347.
Take a look at his solution as it might help you somehow. I'll try your solution for my project by now since it seems simpler. 
Would you mind telling me which board and tool to flash you've been using?

RuhanSA079, 2020-10-07:
The core nrf51822, from waveshare, with a cheapo ST-Link v2. The software used to flash this board is Arduino IDE. I installed all the deps and soft devices. So easy as clicking upload and voila.

RuhanSA079, 2020-10-07:
Closing issue, app timer support not possible at this time.

closed 2020-10-07 by RuhanSA079