sandeepmistry/arduino-nRF5 #430: Support for micro:bit v2

jhmaloney, 2020-10-15:
Do you plan to add support for the new micro:bit (https://microbit.org/new-microbit/)? That uses the nrf52833 chip and it also needs to access pins above pin 31, so a few library changes are needed. If you are interested, I can do PR with the changes I made, although someone should check my changes and also look for additional places that might require the same fix to handle high-numbered pins.

sandeepmistry, 2020-10-18:
Hi @jhmaloney,

No formal plans to add support at this time, however I would be more than happy to review a pull request for this.

cc/ @jaustin

jhmaloney, 2020-10-18:
Great! Besides creating a new variant, I found that a few libraries needed to be extended to handle port B pin numbers (i.e. pin numbers > 31). I will put together what I have so far and issue a PR and you can see what you think. As far as I know, this is still the primary Arduino framework for the micro:bit, so it would be great to make it work with the micro:bit v2.. (Adafruit has a framework for their own nrf5x boards.)

jhmaloney, 2020-10-28:
Submitted PR #432, BBC microbit v2

sandeepmistry, 2020-11-28:
I've updated the nRF MDK and opened a new PR based on @jhmaloney's work: https://github.com/sandeepmistry/arduino-nRF5/pull/439

sandeepmistry, 2020-12-02:
This is in with PR https://github.com/sandeepmistry/arduino-nRF5/pull/439 now!

closed 2020-12-02 by sandeepmistry